davidusb-geek/emhass #9: ERROR:web_server:P_batt was not found in results

purcell-lab, 2022-05-18:
Having issues with the P_batt not being available :-(

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun emhass (no readiness notification)
s6-rc: info: service legacy-services successfully started
[2022-05-19 07:30:45,287] INFO in web_server: Launching the emhass webserver at: http://0.0.0.0:5000
[2022-05-19 07:30:45,287] INFO in web_server: Home Assistant data fetch will be performed using url: http://supervisor/core/api
[2022-05-19 07:30:45,288] INFO in web_server: The base path is: /usr/src
[2022-05-19 07:30:45,292] INFO in web_server: Using core emhass version: 0.3.8
[2022-05-19 07:31:06,375] INFO in command_line: Setting up needed data
INFO:web_server:Setting up needed data
[2022-05-19 07:31:06,431] INFO in forecast: Retrieving weather forecast data using method = scrapper
INFO:web_server:Retrieving weather forecast data using method = scrapper
[2022-05-19 07:31:09,662] INFO in forecast: Retrieving data from hass for load forecast using method = naive
INFO:web_server:Retrieving data from hass for load forecast using method = naive
[2022-05-19 07:31:09,663] INFO in retrieve_hass: Retrieve hass get data method initiated...
INFO:web_server:Retrieve hass get data method initiated...
[2022-05-19 07:31:13,392] INFO in web_server:  >> Performing naive MPC optimization...
INFO:web_server: >> Performing naive MPC optimization...
[2022-05-19 07:31:13,393] INFO in command_line: Performing naive MPC optimization
INFO:web_server:Performing naive MPC optimization
[2022-05-19 07:31:13,416] INFO in optimization: Perform an iteration of a naive MPC controller
INFO:web_server:Perform an iteration of a naive MPC controller
[2022-05-19 07:31:13,563] WARNING in optimization: Failed LP solve with PULP_CBC_CMD solver, falling back to default Pulp
WARNING:web_server:Failed LP solve with PULP_CBC_CMD solver, falling back to default Pulp
[2022-05-19 07:31:13,616] WARNING in optimization: Failed LP solve with default Pulp solver, falling back to GLPK_CMD
WARNING:web_server:Failed LP solve with default Pulp solver, falling back to GLPK_CMD
[2022-05-19 07:31:13,737] INFO in optimization: Status: Optimal
INFO:web_server:Status: Optimal
[2022-05-19 07:31:13,737] INFO in optimization: Total value of the Cost function = -10.1
INFO:web_server:Total value of the Cost function = -10.1
[2022-05-19 07:40:04,528] INFO in command_line: Setting up needed data
INFO:web_server:Setting up needed data
[2022-05-19 07:40:04,539] INFO in web_server:  >> Publishing data...
INFO:web_server: >> Publishing data...
[2022-05-19 07:40:04,540] INFO in command_line: Publishing data to HASS instance
INFO:web_server:Publishing data to HASS instance
[2022-05-19 07:40:04,586] INFO in retrieve_hass: Successfully posted value in existing entity_id
INFO:web_server:Successfully posted value in existing entity_id
[2022-05-19 07:40:04,618] INFO in retrieve_hass: Successfully posted value in existing entity_id
INFO:web_server:Successfully posted value in existing entity_id
[2022-05-19 07:40:04,656] INFO in retrieve_hass: Successfully posted value in existing entity_id
INFO:web_server:Successfully posted value in existing entity_id
[2022-05-19 07:40:04,702] INFO in retrieve_hass: Successfully posted value in existing entity_id
INFO:web_server:Successfully posted value in existing entity_id
[2022-05-19 07:40:04,704] ERROR in command_line: P_batt was not found in results DataFrame. Optimization task may need to be relaunched or it did not converged to a solution.
ERROR:web_server:P_batt was not found in results DataFrame. Optimization task may need to be relaunched or it did not converged to a solution.
```

```yaml
web_ui_url: 0.0.0.0
hass_url: empty
long_lived_token: empty
costfun: profit
optimization_time_step: 30
historic_days_to_retrieve: 2
set_total_pv_sell: false
sensor_power_photovoltaics: sensor.apf_generation_entity
sensor_power_load_no_var_loads: sensor.power_load_no_var_loads
number_of_deferrable_loads: 2
list_nominal_power_of_deferrable_loads:
  - nominal_power_of_deferrable_loads: 1500
  - nominal_power_of_deferrable_loads: 5500
  - nominal_power_of_deferrable_loads: 8000
  - nominal_power_of_deferrable_loads: 2400
list_operating_hours_of_each_deferrable_load:
  - operating_hours_of_each_deferrable_load: 4
  - operating_hours_of_each_deferrable_load: 2
  - operating_hours_of_each_deferrable_load: 2
  - operating_hours_of_each_deferrable_load: 1
list_peak_hours_periods_start_hours:
  - peak_hours_periods_start_hours: '02:54'
  - peak_hours_periods_start_hours: '17:24'
list_peak_hours_periods_end_hours:
  - peak_hours_periods_end_hours: '15:24'
  - peak_hours_periods_end_hours: '20:24'
list_treat_deferrable_load_as_semi_cont:
  - treat_deferrable_load_as_semi_cont: true
  - treat_deferrable_load_as_semi_cont: true
  - treat_deferrable_load_as_semi_cont: false
  - treat_deferrable_load_as_semi_cont: true
load_peak_hours_cost: 0.1907
load_offpeak_hours_cost: 0.1419
photovoltaic_production_sell_price: 0.065
maximum_power_from_grid: 30000
list_pv_module_model:
  - pv_module_model: Advance_Power_API_M370
  - pv_module_model: Advance_Power_API_M370
list_pv_inverter_model:
  - pv_inverter_model: SolarEdge_Technologies_Ltd___SE7600A_US__208V_
  - pv_inverter_model: SolarEdge_Technologies_Ltd___SE7600A_US__208V_
list_surface_tilt:
  - surface_tilt: 18
  - surface_tilt: 10
list_surface_azimuth:
  - surface_azimuth: 90
  - surface_azimuth: 270
list_modules_per_string:
  - modules_per_string: 29
  - modules_per_string: 21
list_strings_per_inverter:
  - strings_per_inverter: 1
  - strings_per_inverter: 1
set_use_battery: true
battery_discharge_power_max: 5000
battery_charge_power_max: 5000
battery_discharge_efficiency: 1
battery_charge_efficiency: 0.95
battery_nominal_energy_capacity: 13500
battery_minimum_state_of_charge: 0.01
battery_maximum_state_of_charge: 0.99
battery_target_state_of_charge: 0.15
nominal_power_of_deferrable_loads: 5500,1500,7000,3000
operating_hours_of_each_deferrable_load: 10,16,3,1
peak_hours_periods_start_hours: 02:54,17:24
peak_hours_periods_end_hours: 15:24,20:24
pv_module_model: Advance_Power_API_M370,Advance_Power_API_M370
pv_inverter_model: >-
  SolarEdge_Technologies_Ltd___SE7600A_US__208V_,SolarEdge_Technologies_Ltd___SE7600A_US__208V_
surface_tilt: 10,10
surface_azimuth: 90,270
modules_per_string: 29,21
strings_per_inverter: 1,1
```

davidusb-geek, 2022-05-19:
This is normally solver in the latest version. The filename to retrieve the data when publishing was simply wrong. I think that this was directly related to this issue.

purcell-lab, 2022-05-19:
Thank you, seems stable after all your work in the last day.

closed 2022-05-19 by purcell-lab