davidusb-geek/emhass #11: FR: Pass additional sensor entities.

purcell-lab, 2022-06-07:
Working quite well here now and stable.

Would it be possible to pass some additional sensor entities to HA?

p_grid_forecast & total value of cost function 

Thanks.

My optimisation for next 24 hrs, total value of cost function $63:

[image]

davidusb-geek, 2022-06-07:
Ok, closing the issue on the add-on repository then. 

Wow after 5pm it will literally be off the charts! And a complete battery capacity needed.

Yes these new sensor entities can be also added. I will do that on the next version.

davidusb-geek, 2022-06-10:
Hi, I just implemented this on the latest version >> v0.3.18

purcell-lab, 2022-06-13:
Thanks,

Working well.

```
/usr/local/lib/python3.9/dist-packages/emhass/web_server.py:46: FutureWarning:
Dropping of nuisance columns in DataFrame reductions (with 'numeric_only=None') is deprecated; in a future version this will raise TypeError.  Select only valid columns before calling the reduction.
[2022-06-13 11:14:06,857] INFO in command_line: Setting up needed data
[2022-06-13 11:14:06,869] INFO in web_server:  >> Publishing data...
[2022-06-13 11:14:06,870] INFO in command_line: Publishing data to HASS instance
[2022-06-13 11:14:06,948] INFO in retrieve_hass: Successfully posted to sensor.p_pv_forecast = 9644.92
[2022-06-13 11:14:06,994] INFO in retrieve_hass: Successfully posted to sensor.p_load_forecast = 464.07
[2022-06-13 11:14:07,036] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable0 = 1500.0
[2022-06-13 11:14:07,075] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable1 = 5500.0
[2022-06-13 11:14:07,113] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable2 = 0.0
[2022-06-13 11:14:07,157] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable3 = 0.0
[2022-06-13 11:14:07,199] INFO in retrieve_hass: Successfully posted to sensor.p_batt_forecast = -2180.85
[2022-06-13 11:14:07,245] INFO in retrieve_hass: Successfully posted to sensor.soc_batt_forecast = 77.67
[2022-06-13 11:14:07,309] INFO in retrieve_hass: Successfully posted to sensor.p_grid_forecast = 0.0
[2022-06-13 11:14:07,364] INFO in retrieve_hass: Successfully posted to sensor.total_cost_fun_value = -4.54
[2022-06-13 11:15:00,081] INFO in command_line: Setting up needed data
[2022-06-13 11:15:00,085] INFO in retrieve_hass: Retrieve hass get data method initiated...
[2022-06-13 11:15:00,953] INFO in forecast: Retrieving weather forecast data using method = scrapper
[2022-06-13 11:15:04,654] INFO in forecast: Retrieving data from hass for load forecast using method = naive
[2022-06-13 11:15:04,656] INFO in retrieve_hass: Retrieve hass get data method initiated...
[2022-06-13 11:15:06,970] INFO in web_server:  >> Performing naive MPC optimization...
[2022-06-13 11:15:06,970] INFO in command_line: Performing naive MPC optimization
[2022-06-13 11:15:06,978] INFO in optimization: Perform an iteration of a naive MPC controller
[2022-06-13 11:15:07,270] INFO in optimization: Status: Optimal
[2022-06-13 11:15:07,270] INFO in optimization: Total value of the Cost function = -4.54
```