davidusb-geek/emhass #13: Error on dayahead-optim POST call

vdiogo, 2022-06-19:
Hi

I'm getting an error when triggering the REST call for the dayahead-optim.

This is what I get in the logs:
```
[2022-06-19 10:45:17,391] INFO in command_line: Setting up needed data
[2022-06-19 10:45:17,399] INFO in forecast: Retrieving weather forecast data using method = scrapper
[2022-06-19 10:45:19,945] INFO in forecast: Retrieving data from hass for load forecast using method = naive
[2022-06-19 10:45:19,947] INFO in retrieve_hass: Retrieve hass get data method initiated...
[2022-06-19 10:45:47,744] INFO in web_server:  >> Performing dayahead optimization...
[2022-06-19 10:45:47,745] INFO in command_line: Performing day-ahead forecast optimization
[2022-06-19 10:45:47,755] INFO in optimization: Perform optimization for the day-ahead
[2022-06-19 10:45:47,894] ERROR in optimization: It was not possible to find a valid solver for Pulp package
[2022-06-19 10:45:47,895] INFO in optimization: Status: Not Solved
[2022-06-19 10:45:47,895] WARNING in optimization: Cost function cannot be evaluated, probably None
[2022-06-19 10:45:47,904] ERROR in app: Exception on /action/dayahead-optim [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 151, in action_call
    opt_res = dayahead_forecast_optim(input_data_dict, app.logger)
  File "/usr/local/lib/python3.9/dist-packages/emhass/command_line.py", line 195, in dayahead_forecast_optim
    opt_res_dayahead = input_data_dict['opt'].perform_dayahead_forecast_optim(
  File "/usr/local/lib/python3.9/dist-packages/emhass/optimization.py", line 493, in perform_dayahead_forecast_optim
    self.opt_res = self.perform_optimization(df_input_data, P_PV.values.ravel(),
  File "/usr/local/lib/python3.9/dist-packages/emhass/optimization.py", line 384, in perform_optimization
    opt_tp["P_grid"] = [P_grid_pos[i].varValue + P_grid_neg[i].varValue for i in set_I]
  File "/usr/local/lib/python3.9/dist-packages/emhass/optimization.py", line 384, in <listcomp>
    opt_tp["P_grid"] = [P_grid_pos[i].varValue + P_grid_neg[i].varValue for i in set_I]
TypeError: unsupported operand type(s) for +: 'NoneType' and 'NoneType'
```

This is my config.yaml configuration:
```
rest_command:
  dayahead_optim: 
    url: http://192.168.86.29:5000/action/dayahead-optim
    method: POST
    content_type: 'application/json'
    payload: '{}'

  publish_data: 
    url: http://192.168.86.29:5000/action/publish-data
    method: POST
    content_type: 'application/json'
    payload: '{}'
```

davidusb-geek, 2022-06-19:
Hi, the system is not finding a valid solver.

Try changing the solver in the configuration to `COIN_CMD` and pass the path `/usr/bin/cbc`.

As a second option you could try to use the alternative GPLK solver, you will need to use `GLPK_CMD` as the solver and maybe manually install the GLPK solver with:

```
sudo apt-get install glpk-utils
```

You will need to to this inside the container. To go inside the container do:
```
docker exec -it <container-name> /bin/bash
```
The container name can be found using:
```
docker ps -a
```

davidusb-geek, 2022-06-21:
Hi, closing this for now as this is a know and proven solution for this. Again feel free to reopen if you need assistance.