davidusb-geek/emhass #16: Passing forecast data do not work

overas, 2022-08-29:


overas, 2022-08-29:
Yes I have refreshed the webpage after triggering the service "Shell Command: load_cost_forecast". The page do not update using Chrome or with Firefox.

davidusb-geek, 2022-08-31:
Hi, the problem persists after updating to the latest version?

overas, 2022-08-31:
Yes, the problem is the same with EMHASS add-on v0.2.20

overas, 2022-09-02:
I solved it! I found out I have done two errors. 

The first was passing wrong amount of data in this list. Nordpool publish prices for every hour so the list must have 24 data points, not 48 points.
`You need to be careful here to send the correct amount of data on this list, the correct length. For example, if the data time step is defined to 1h and you are performing a day-ahead optimization, then this list length should be of 24 data points.`

The second was using template with errors. My template was wrong with the parenthesis and curly bracket. After fixing the template the shell_command worked and the passing of load_cost_forecast and prod_price_forecast was successful.

```
shell_command:
  publish_data: "curl -i -H 'Content-Type:application/json' -X POST -d '{}' http://localhost:5000/action/publish-data"
  
  post_nordpool_forecast: "curl -i -H 'Content-Type: application/json' -X POST -d '{\"load_cost_forecast\":{{(
        (state_attr('sensor.nordpool_euro', 'raw_today')|map(attribute='value')|list)[:24])
        }},\"prod_price_forecast\":{{(
        (state_attr('sensor.nordpool_euro', 'raw_today')|map(attribute='value')|list)[:24])}}}' http://localhost:5000/action/dayahead-optim"
```

Question:
1)
```
Load cost forecast: the price of the energy from the grid on the next 24h. This is given in EUR/kWh.
PV production selling price forecast: at what price are you selling your excess PV production on the next 24h. This is given in EUR/kWh.
```
Can I use NOK (norwegian) as price instead of Euro?

overas, 2022-09-02:
[image]

davidusb-geek, 2022-09-02:
Ok perfect!
I'm glad that you find the problem. 
Yes of course you can use whatever currency you want as long as you use the same currency everywhere, no problem at all.

closed 2022-09-02 by davidusb-geek