davidusb-geek/emhass #18: Document setup for native solcast forecast module

purcell-lab, 2022-08-31:
I couldn't find the documentation for setting up the native solcast module.

I gather from https://github.com/davidusb-geek/emhass/commit/b22b4a867a81ff21e00d00c82af2cb4b1e624292, that I need to update an secrets_emhass.yaml file, but I couldn't locate that.  Is it inside the docker image or elsewhere?

Can I set these parameters from the normal emhass configuration file?

solcast_api_key: yoursecretsolcastapikey
solcast_rooftop_id: yourrooftopid

davidusb-geek, 2022-08-31:
Oh yes! This is still missing. I have just added support for Solcast on the core EMHASS module, now I just need to add the correct configuration on the add-on side. I will be doing this anytime soon

davidusb-geek, 2022-09-15:
Hi, it is small but functional documentation. SolCast can now be used natively on emhass.
Follow these instructions: https://emhass.readthedocs.io/en/latest/forecasts.html#pv-power-production-forecast

closed 2022-09-15 by davidusb-geek