davidusb-geek/emhass #19: Adding real examples in the documentation and better explain parameters/abbreviations

daman7cz, 2022-09-03:
Hello davidusb-geek,
Thank you for your project you have created.

Concerning the settings and documentation - I am struggling for several days to start the addon correctly without having errors in log. Even though I have read trough all notes, I still had issues. Thus I would propose to have more verbose StartUp Guide. I know that documentation was always the least loved part for me, but if we see real example (exact wording)  it would help a lot :-)

The addon is rather sensitive for correct fillings (IP address with or without http prefix, or not allowing "localhost" word, but allowing IP address 0.0.0.0, stating http or https, etc). 

Also in the next steps, for example when I was hesitating if to optimize for cost or profit, the difference was not clear to me. And some of the abbreviations were not clear too.

Thank you for cosidering it and thank you again for your addon.
Daman

davidusb-geek, 2022-09-03:
Thank you!
Yes of course there will always be room to improve that documentation.
I'll put this in the to do list

davidusb-geek, 2022-10-13:
Added more examples and a translation file for the add-on.

closed 2022-10-13 by davidusb-geek