davidusb-geek/emhass #37: Combining with thresholds

Drazhar, 2023-01-16:
Hey there

First of all, you got a nice project here! I was just thinking to implement something similar in an HASS integration. But then I googled before I start such a big undertaking and found your Project 👍

If I understand it correctly, you calculate the optimal device load times based on the PV forecast and expected non controllable loads based on the past by solving a LP problem. My idea was quite similar, but instead of solving for the optimal times to turn the devices on, I wanted to solve for the optimal threshold for the devices to turn on.

Currently my biggest consumers which I control are:
- EV
- Conduction heater for water

Also I try to estimate when to turn on the dish washer and tumbler/washing machine.
This works quite well using a set threshold for the conduction heater. The EV charges dependent on the PV power with an factor how much grid/PV power can be used. The only downside is defining the thresholds. This is currently manual guess work based on my PV forecast (which is an self made implementation which uses a linear regression model based on my historic production and weather. For me it's more precise than solcast or similar I have tried).

Do you think it's feasible to implement a threshold prediction into your system? I haven't digged trough the code yet, but I think it should be possible. It would be great to define different type of devices:

- Time based control
- Threshold based
- Combined (for example during the day threshold based, but if a specific minimum power is required, it could be time based during night. For example if the EV has to have >70% SOC at the morning)

Also for my water heater I know that it doesn't have to heat every day, but at least 10kW in 3 days. This would also be great to implement into the LP.

If you think this features make sense in your project, I'm happy to contribute to the code! We could also implement the PV forecast based on the historic data, but this would most likely make more sense in a separate integration.

Best regards
Philip

davidusb-geek, 2023-01-16:
Hi and thanks!

> If I understand it correctly, you calculate the optimal device load times based on the PV forecast and expected non controllable loads based on the past by solving a LP problem. My idea was quite similar, 

Yes it basically give you the optimal schedule for your deferrable/controlable loads.

> Do you think it's feasible to implement a threshold prediction into your system? I haven't digged trough the code yet, but I think it should be possible.

Yes everything is possible. However since I began with this code my vision was for controling based on thresholds and other heuristics it is better to do that kind of tasks directly on Home Assistant and taking advantage of the power of automations there.

However if you want to propose some improvements to the actual code pull requests are more than welcomed. For new ideas like these may be a proof of concept will be nice and on those results then try to integrate that into the existing code may be an option.

And goal however is to minimize or avoid at all any braking changes.

> Also for my water heater I know that it doesn't have to heat every day, but at least 10kW in 3 days. This would also be great to implement into the LP.

Did you meant 10 kWh?
That type of action can be achieved with templating and live playing the optimization parameters directly from HA. There are some examples on the documentation to achieve this.

> We could also implement the PV forecast based on the historic data, but this would most likely make more sense in a separate integration.

Forecast based on history data and known future data (like the timestamps) is a work in progress. However I'm only working on this for load forecasting as PV forecasts are quite precise using the PVLib or any of the simplified web service methods like Solcast or Solar.Forecast.

I think that it may be ok to include forecast models inside this EMHASS module if the models are fairly simple as linear regression and the such. Even more complicated model as decision trees or ensemble method may be considered. But definetly not deep learning, those are for a separate integration.

Drazhar, 2023-01-16:
Great detailed answer!

> However since I began with this code my vision was that for controling based on thresholds and other heuristics it is better to do that kind of tasks directly on Home Assistant and taking advantage of the power of automations there.

Yes, I control everything with thresholds and automations and it works quite good. Maybe I'm not using the best solution, since I recently switched from OpenHAB to HomeAssistant for multiple reasons. But I have to define the thresholds manually right now. As an easy example: I know that the car won't be home during a short, cloudy winter day and the boiler need to run for at least 3 hours with 2 kW (6kWh). Based on my forecast I know, that I have 3 hours where my PV production is above 1.5 kW. This means I can set the threshold to 1.3-1.5 and be quite sure that I have enough warm water at the evening.

This becomes more complicated when the car is at home and also needs 10-20 kWh for the next day, the washing machine has to run and maybe someday I will get a heat pump. Then my idea was to use LP to calculate the ideal times of the day when everything should run (for which your EMHASS seems to be perfect) and then based on the results I can extract thresholds for the things I can turn on and off by demand (eg. boiler and EV), the rest can run by time. Using thresholds instead of fixed times, I can compensate for incorrect forecasts and heat/charge more if there should be more sun then predicted compared to fixed times. Of cause then the LP has to rerun, but LPs are quite cheap, so it shouldn't be an issue, even on a rpi.

Maybe this can also be achieved already with EMHASS and some tweaking. I still have to migrate some stuff from Openhab to HASS, after that I have to test it with EMHASS :-)

> Did you meant 10 kWh?

Yes of cause, sorry.

> I think that it may be ok to include forecast models inside this EMHASS module if the models are fairly simple as linear regression and the such.

Yes, it's a simple linear regression model and therefore even on an rpi extremely cheap. Only the training is a bit more expensive, this I currently do on my PC and just transfer the model to the rpi. There it's also just expensive because I make a grid search cv to choose the best combination of weather data and polynomals. Without, it would also be quite cheap.

Hopefully during this week I find some time to finish my migration and test EMHASS on my system. Then I can come back to you with some more sophisticated proposals.

davidusb-geek, 2023-05-14:
Since this conversation a new class containing a forecasting methodology using machine learning models was introduced in the latests releases.
I will be closing this issue but if you want to further discuss the threshold idea then feel free to reopen. 
As I said for me this kind of task should be done directly on HA automations. But I'm open to discuss any improvements to the current code. For any new proposition like this some proof of concept could be nice to advance in the discussion.

closed 2023-05-14 by davidusb-geek