davidusb-geek/emhass #40: 0.3.23 AttributeError: module 'numpy' has no attribute 'typeDict'

purcell-lab, 2023-01-27:
**Describe the bug**
0.3.23 fails to start.

```
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 14, in <module>
    from emhass.command_line import set_input_data_dict
  File "/usr/local/lib/python3.9/dist-packages/emhass/command_line.py", line 13, in <module>
    from emhass.forecast import forecast
  File "/usr/local/lib/python3.9/dist-packages/emhass/forecast.py", line 10, in <module>
    import pvlib
  File "/usr/local/lib/python3.9/dist-packages/pvlib/__init__.py", line 3, in <module>
    from pvlib import (  # noqa: F401
  File "/usr/local/lib/python3.9/dist-packages/pvlib/clearsky.py", line 14, in <module>
    import h5py
  File "/usr/lib/python3/dist-packages/h5py/__init__.py", line 21, in <module>
    from . import _debian_h5py_serial as _h5py
  File "/usr/lib/python3/dist-packages/h5py/_debian_h5py_serial/__init__.py", line 46, in <module>
    from ._conv import register_converters as _register_converters
  File "h5py/_debian_h5py_serial/_conv.pyx", line 1, in init h5py._debian_h5py_serial._conv
  File "h5py/_debian_h5py_serial/h5t.pyx", line 293, in init h5py._debian_h5py_serial.h5t
  File "/usr/local/lib/python3.9/dist-packages/numpy/__init__.py", line 284, in __getattr__
    raise AttributeError("module {!r} has no attribute "
AttributeError: module 'numpy' has no attribute 'typeDict'
```

**To Reproduce**
Upgraded to 0.3.23

**Expected behavior**
Functional as per past version

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Home Assistant installation type**

 - Home Assistant Supervised

**Your hardware**
- OS: Debian
- Architecture:  aarch64

**EMHASS installation type**
 - Add-on,
 
**Additional context**

purcell-lab, 2023-01-27:
Rolled back to addon_5b918bf2_emhass_0.2.23, restarted add-on and functional with previous version.

kcoffau, 2023-01-27:
Same issue here

davidusb-geek, 2023-01-27:
I'm currently working on a solution to this issue.

davidusb-geek, 2023-01-27:
Issues should be normally solved, images are being built.

purcell-lab, 2023-01-27:
Fixed in 0.[2,3].25.

Thanks for the quick turnaround.

closed 2023-01-27 by purcell-lab

kcoffau, 2023-01-27:
Wow! Amazing David! Thank you

lovekull76, 2023-01-28:
Great!! Thanks David, I’ll check it out!