davidusb-geek/emhass #41: Feature Request: more verbose logging

purcell-lab, 2023-01-28:
**Describe the bug**
It would be great to get more verbose logging during error conditions.

**To Reproduce**

From my logs, I am passing some invalid data, but I don't know what it is.

```logs
[2023-01-28 14:10:00,382] INFO in command_line: Setting up needed data
[2023-01-28 14:10:00,384] WARNING in utils: There are non numeric values on the passed data, check for missing values (nans, null, etc)
[2023-01-28 14:10:00,385] WARNING in utils: There are non numeric values on the passed data, check for missing values (nans, null, etc)
[2023-01-28 14:10:00,388] INFO in retrieve_hass: Retrieve hass get data method initiated...
```

**Expected behavior**
It would be useful to include the out of bounds data in the logs.

**Home Assistant installation type**
 
 - Home Assistant Supervised
 

**Your hardware**
-OS: Linux, etc
- Architecture: aarch64

**EMHASS installation type**
 - Add-on

**Additional context**
Add any other context about the problem here.

purcell-lab, 2023-01-28:
I'm struggling to find the non numeric values here:

```bash
curl -i -H "Content-Type: application/json" -X POST -d '{
 "load_cost_forecast":[0.22, 0.21, 0.23, 0.25, 0.24, 0.27, 0.31, 0.42, 0.32, 0.31, 0.28, 0.31, 0.26, 0.28, 0.29, 0.27, 0.25, 0.23, 0.25, 0.24, 0.24, 0.23, 0.23, 0.23, 0.22, 0.22, 0.22, 0.22, 0.22, 0.2, 0.19, 0.16, 0.15, 0.14, 0.12, 0.12, 0.15, 0.15, 0.15, 0.15, 0.15, 0.2, 0.2, 0.2, 0.22, 0.22, 0.24, 0.21], 
 "prod_price_forecast":[0.13, 0.12, 0.14, 0.16, 0.15, 0.18, 0.21, 0.31, 0.22, 0.22, 0.18, 0.21, 0.17, 0.18, 0.19, 0.17, 0.15, 0.13, 0.15, 0.14, 0.14, 0.13, 0.13, 0.13, 0.13, 0.13, 0.12, 0.12, 0.12, 0.11, 0.09, 0.07, 0.07, 0.06, 0.04, 0.04, 0.07, 0.06, 0.06, 0.06, 0.07, 0.11, 0.11, 0.12, 0.13, 0.13, 0.15, 0.12], 
 "pv_power_forecast":[5883, 9816, 7968, 5973, 4029, 2048, 412, 11, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 11, 415, 1590, 3203, 4859, 6555, 8292, 9924, 11556, 12746, 13262, 13925, 14326, 14796, 14878, 14790, 14500, 14113, 13947, 13474], 
 "prediction_horizon":48,
 "soc_init":0.97,
 "def_total_hours":[0,0,0,0,0]}'
 http://localhost:5000/action/naive-mpc-optim
```

davidusb-geek, 2023-01-29:
I've just released a new version addressing this. The message are now really more explicit and also there was a bug, I wrongly detecting numeric values as non-numeric.

purcell-lab, 2023-01-29:
Thanks for the quick turn around.

closed 2023-01-29 by purcell-lab