davidusb-geek/emhass #42: Time Zone not taken from secrets.yaml in Docker

smitterer, 2023-01-28:
**Describe the bug**
The time zone settintgs do not seem to be taken from the secrets.yaml when using the docker image from docker hub. If I manually add it to the docker parameters, it is working.

**To Reproduce**
Try it with only setting it in the yaml and then when also placing it into the docker parameters.

**Expected behavior**
Take timezone from yaml.

**Home Assistant installation type**
 - Home Assistant Core on Synology Docker

**Your hardware**
- OS: Synology

**EMHASS installation type**
 - Docker from hub

davidusb-geek, 2023-01-28:
Thanks for reporting this. Could you please tell me what version of emhass you are using?

davidusb-geek, 2023-01-28:
And just to be sure, are you passing the secrets data yaml file when launching the container like this?:
```
docker run -it --restart always -p 5000:5000 -e "LOCAL_COSTFUN=profit" -v $(pwd)/config_emhass.yaml:/app/config_emhass.yaml -v $(pwd)/secrets_emhass.yaml:/app/secrets_emhass.yaml --name DockerEMHASS <REPOSITORY:TAG>
```

smitterer, 2023-01-29:
I'm using 0.3.25.
The containers are started via the Synology GUI but the files are mapped, yes. Also the deferrable sensor values from EMHASS to HA are correctly transferred so I gues this tells us that it reads the yaml files.

davidusb-geek, 2023-01-29:
I cannot see in the code how if present in the yaml file the time zone would not be passed later to the computations.
The only thing I can see for now is the presence of `time_zone: Europe/Paris` respecting this format. Did you double check that?

smitterer, 2023-01-30:
time_zone: Europe/Vienna is my setting. I just installed the newest version of EMHASS and now do not have the problem, also after removing the TZ value from the parameters. So I guess something went wrong while running the previous one.

smitterer, 2023-01-30:
Ups sorry, just reloaded the page and time is wrong again. So the issue is still there.

davidusb-geek, 2023-01-30:
Ok, there has been some modifications lately. Please upgrade to the latest v0.3.34 and see if there any improvements. You should be able to pull the latest image very soon. Just wait for v0.3.34 to be availabe on dockerhub

davidusb-geek, 2023-02-05:
Hi, did you had the time to test this with the latest version?

davidusb-geek, 2023-02-28:
Feel free to reopen if this wasn't solved with latest version.

closed 2023-02-28 by davidusb-geek