davidusb-geek/emhass #48: Config file uses arrays of single-element dicts: ??

smurfix, 2023-02-11:
I wonder what your motivation is to use arrays of single-element dicts virtually everywhere in your configuration file.
IMHO that's unnecessarily complex. Would you accept a patch that drops this?

davidusb-geek, 2023-02-11:
Yes totally agree with you. This was the first solution I implemented but is awful and over complex as you said.
If you can help with this it would be appreciated. Just be careful to test the web_server.py after the changes because this file is not unit tested.

smurfix, 2023-02-11:
Working on it. :-)

davidusb-geek, 2023-02-14:
Hi, did you managed to test your changes? It works as expected?

smurfix, 2023-02-14:
Not yet, too much other stuff on my plate at the moment. Coming soon.

davidusb-geek, 2023-02-15:
Sure no problem, just asking because I will be adding a bunch of new stuff soon, so just looking to avoid code conflicts. But if any it should not be so hard to solve with git.

davidusb-geek, 2023-02-22:
If you know how to unittest the web_server.py that uses Flask, any advice will be greatly appreciated.

davidusb-geek, 2024-01-29:
Closing as completed #69

closed 2024-01-29 by davidusb-geek