davidusb-geek/emhass #56: Feature request: Let users change the default entity name "sensor.p_deferrableX" to a friendly name

overas, 2023-02-21:
I has been nice if users can change the default entity name  "sensor.p_deferrableX" to for instance the friendly name "sensor.heating_cable_kitchen". I think it can be done if the entity has a unique ID. 

`This entity ('sensor.p_deferrable0') does not have a unique ID, therefore its settings cannot be managed from the UI. See the [documentation](https://www.home-assistant.io/faq/unique_id) for more detail.`

davidusb-geek, 2023-03-07:
This was added in the latest release.

When using the `publish-data` endpoint just add some runtime parameters like this:
```
shell_command:
  publish_data: "curl -i -H \"Content-Type:application/json\" -X POST -d '{\"custom_load_forecast_id\": {\"entity_id\": \"sensor.p_load_forecast\", \"unit_of_measurement\": \"W\", \"friendly_name\": \"Load Power Forecast\"}}' http://localhost:5000/action/publish-data"
```

These keys are available to modify: `custom_pv_forecast_id`, `custom_load_forecast_id`, `custom_batt_forecast_id`, `custom_batt_soc_forecast_id`,  `custom_grid_forecast_id`, `custom_cost_fun_id`, `custom_deferrable_forecast_id`.

If you provide the `custom_deferrable_forecast_id` then the passed data should be a list of dictionaries, like this:
```
shell_command:
  publish_data: "curl -i -H \"Content-Type:application/json\" -X POST -d '{\"custom_deferrable_forecast_id\": [{\"entity_id\": \"sensor.p_deferrable0\",\"unit_of_measurement\": \"W\", \"friendly_name\": \"Deferrable Load 0\"},{\"entity_id\": \"sensor.p_deferrable1\",\"unit_of_measurement\": \"W\", \"friendly_name\": \"Deferrable Load 1\"}]}' http://localhost:5000/action/publish-data"
```
And you should be careful that the list of dictionaries has the correct length, which is the number of defined deferrable loads.

closed 2023-03-07 by davidusb-geek

overas, 2023-03-07:
Thanks. I am going to try the new friendly name in emhass 3.0.1.