davidusb-geek/emhass #58: Feature Request: Provide all senors / columns in Home Assistant

smitterer, 2023-03-03:
**Describe the bug**
As described under https://community.home-assistant.io/t/emhass-an-energy-management-for-home-assistant/338126/445?u=smi it would be awesome to have all senors / columns shown in EMHASS as sensors in Home Assistant. So not only those which are available currently (like defferable loads), but also unit_load_cost and unit_prod_price.

**To Reproduce**
None

**Expected behavior**
No issue, just a request

**Screenshots**
not applicable

**Home Assistant installation type**
Applies to all

**Your hardware**
Applies to all

**EMHASS installation type**
Applies to all

**Additional context**
None

davidusb-geek, 2023-03-03:
Yes it will be done. I will just give priority to the current developments (a better load forecasting model) and then I'll look for these improvements.

davidusb-geek, 2023-05-14:
Hi added this in the latest commit 8a23a35
This will be available soon in the next release

closed 2023-05-14 by davidusb-geek