davidusb-geek/emhass #66: How to create a deferrable load sensor

vdiogo, 2023-03-15:
First off, great project. Can't wait to start using/ benefiting from it

I need to create a new power sensor which picks up on my consumed power (A: the house electricity consumption) and deducts the power consumed by the device I want to deduct (B: boiler thermostats resistance).

I have the two (A and B) and can build a template sensor in Home Assistant, BUT A returns the delta of value effectively consumed by the house, meaning, during PV generation, this value can become negative, as the house is spending less than what the solar panels are generating.

What is the right way to go here for the new sensor that I need to add in EMHASS for the load power without the deferrable loads? Just a subtraction (A-B)? Or something else?

davidusb-geek, 2023-03-15:
Hi. We need the power consumption which is not affected in any way by your PV production. So A-B won't be good by any means. 
So, two options:
1) If you have a sensor C with your PV production, the just do A+C
2) Install a current clamp of an energy meter for your power consumption.

vdiogo, 2023-03-15:
But of course, you are right! Thanks

closed 2023-03-16 by davidusb-geek