davidusb-geek/emhass #71: Feature Request: smooth allocation of power to continuous loads

purcell-lab, 2023-04-10:
I have three continuous loads in my configuration.

P_deferrable2 (EV charger) can take values between 0W to + 11 kW
P_deferrable3 (HVAC) can take values between 0W to 4 kW
P_batt which can take values from -15 kW to + 15 kW

[image]

As you can see in the above graph, the allocation of power to these loads is very lumpy between time slots.

For example rather than having the EV charging; [21530,11500,2173,11500] could this be smoothed to [6831, 6831, 6831, 6831] so the same in aggregate?

Similar for battery charging currently [0, -15000, 0, -15000] could this be smoothed to [-7500, -7500, -7500, -7500]?

[image]

The benefits of a smoothed allocation is all continuous loads receive an ongoing proportion rather than starting and stopping on alternative time slots (useful for battery and HVAC) and if things change, such as the EV departing, it will have received some power allocation earlier in the day.

davidusb-geek, 2023-05-21:
This is a very important improvement for me but I just haven't been able to find the time to work on this lately. But it is in the to do list.

davidusb-geek, 2023-05-27:
This is implemented in the latest release that I've just launched. It should be available soon.
I did some testing on this. The main thing is that it is hard to add additional constraints to the deferrable loads because there is a main power balance contraint. Adding new constraints to smooth the power then almost always resulted in infeasible conditions.
So what I did to solve this problem is just to add this dynamic constraint to the battery power. In my tests it seems to work very well to smooth both the battery power and the deferrable load power at the same time. But this still needs more testing on your side to see if its working as intended. I guess that this makes senses as this is why the battery is there for, to add a much desired flexibility right?
The downside is for people that don't have a battery (like me), where I just don't see it possible to add more contraints to the deferrable loads without affecting the LP feasibility.
Please test and give your feedback when possible.
Cheers

purcell-lab, 2023-05-28:
Looking forward to testing.  Thanks.

purcell-lab, 2023-05-31:
I am wondering if it is possible for a continuous load to also be defined as having just one startup (set_def_constant).

Deferrable3 - HVAC in this case is still starting and stopping, even with the battery smoothing.

If it could be constrained to have only one run window, then it should present a smoothed outcome, rather than starting and stopping.

Is this the correct payload to achieve this?

```yaml
        "num_def_loads": 6,
        "def_total_hours": {{[states('sensor.def_total_hours_pool_filter')|int(0),
                              states('sensor.def_total_hours_pool_heatpump')|int(0),
                              states('sensor.def_total_hours_ev')|int(0),
                              states('sensor.def_total_hours_hvac')|int(0),
                              states('sensor.def_total_hours_hws')|int(0),
                              states('input_number.car2_def_total_hours')|int(0)]}}
        "P_deferrable_nom": [1340, 5000, 11500, 4000, 600,11500],
        "treat_def_as_semi_cont": [1, 1, 0, 0, 1, 0],
        "set_def_constant":       [0, 0, 1, 1, 0, 1]
```

[image]

davidusb-geek, 2023-05-31:
> I am wondering if it is possible for a continuous load to also be defined as having just one startup (set_def_constant).
 
Yes you can try this I don't see why not. It is converging to a feasible solution?
Your payload seems ok to me.

purcell-lab, 2023-08-11:
The battery smoothing from these maximum power dynamic settings are working well to smooth the battery rate of change.

[image]

It is has shifted the peaks into my p_grid forecasts. 1900 is with battery smoothing, but 0900 is without battery smoothing.

[image]

p_deferrable5 is my EV charging which is a continuous load, but just switches between peaks.

[image]

I'm wondering if we could have an option to set the dynamic power constraints on p_grid, if that would result in a feasible solution and smooth out these peaks?

davidusb-geek, 2023-09-15:
> I'm wondering if we could have an option to set the dynamic power constraints on p_grid, if that would result in a feasible solution and smooth out these peaks?

How in any case it would be interesting to add a dynamic constraint to the grid power? 
That power should be let free as it is the variable that gives the most important degree of freedom. Adding a constraint to it will certainly lead to lots of infeasible scenarios.

PieterVO, 2023-11-06:
> How in any case it would be interesting to add a dynamic constraint to the grid power? That power should be let free as it is the variable that gives the most important degree of freedom. Adding a [...]

In Belgium there is an additional cost if you have power peaks over 2.5 kw (measured per 15 minutes).
This only counts for consumption, not injection.
Keeping energy consumption from grid below 2.5 kw (or any set value) can bring significant cost savings vs letting the power reach the maximum possible.
Limiting the grid power to 2.5 kw with the current configuration variable also impacts the maximum injection power, which is undesired.

Is there any way we can easily add this kind of logic where the power from grid is limited to a certain level?

Most inverters can currently go in some "Peak Shaving" mode, where it uses the battery that it has left to level off any peaks you may have above a certain level.

closed 2025-01-31 by davidusb-geek