davidusb-geek/emhass #73: Feature Request: Use entity_id's instead of P_deferrable_nom

gieljnssns, 2023-05-10:
If you have a powermeter on the deferrable load why can't you use it in the configuration?

davidusb-geek, 2023-05-10:
I don't fully understand your request. Could you please elaborate a little bit more?

A real measurement on your deferrable load using a power meter is actually used when you provide the previous historical values of your home consumption minus that measurement of the deferrable load power. So that data is used for the optimization.
Then `P_deferrable_nom` is just a parameter definition to have the maximum possible value of a deferrable load so that the optimized power don't exceed that value.
I don't if this is what you meant or if it was something else?

gieljnssns, 2023-05-10:
Thanks for your answer.
I have some more questions.

- I know you can change the entity_id of the `sensor.pdeferrable`, but can't you make it configurable in the add-on or in the webserver?
- I use the solcast integration in HA, why can't we pass that sensor to EMHASS

purcell-lab, 2023-05-10:
Regarding Solcast, I also use that integration.

I agree it would be useful if EMHASS could just lookup that sensor and use the values directly. Rather than making addition calls on the SolCast API.

I extract the values from the SolCast integration and inject them into EMHASS using this template:

```yaml
            + state_attr('sensor.solcast_forecast_today', 'detailedForecast')|selectattr('period_start','gt',utcnow())
              | map(attribute='pv_estimate')|map('multiply',2000)|map('int')|list 
            + state_attr('sensor.solcast_forecast_tomorrow', 'detailedForecast')|selectattr('period_start','gt',utcnow())
              | map(attribute='pv_estimate')|map('multiply',2000)|map('int')|list
 ```

gieljnssns, 2023-05-10:
@purcell-lab 
Do you have your whole HA configuration somewhere on github?
So I can use parts of your emhass configuration.

davidusb-geek, 2023-05-10:
> * I know you can change the entity_id of the `sensor.pdeferrable`, but can't you make it configurable in the add-on or in the webserver?

For now it is configurable on the shell command for publish data. 

> * I use the solcast integration in HA, why can't we pass that sensor to EMHASS

You can pass any forecast data data that you want using the `pv_power_forecast` key in your shell command along with templates as the one shown by @purcell-lab 

It is not planned to do anymore than this as it is fully functional and this approach allows for a fairly generalized code.

gieljnssns, 2023-05-11:
Ok, thanks for your answers.

I just wan't to mention, 
I know a little bit from coding and stuff and I find EMHASS difficult to get it running.
I think when it was easier to configure a lot more people can use this addon.
If someone who has this running on a daily used setup could share his whole config, people could some real use cases.

purcell-lab, 2023-05-27:
> Another question. Is it possible to define a `minimum time on` for a load? In my heat pump for hot water is a timer relais built in. So when I toggle it on it takes 25min to start the heat pump.

I put delays for minimum time in my automations.

[image]