davidusb-geek/emhass #75: UnicodeDecodeError

jahpster, 2023-05-17:
**Describe the bug**I
When I installed your addon it worked for a few hours but then I got an error while running emhass ass addon on ha supervisor.

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun emhass (no readiness notification)
s6-rc: info: service legacy-services successfully started
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 285, in <module>
    injection_dict = pickle.load(fid)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x95 in position 3591584: invalid start byte

**Home Assistant installation type**
 
 - Home Assistant Supervised
 

**Your hardware**
- OS: HA OS
- Architecture: amd64

**EMHASS installation type**
 - Add-on

davidusb-geek, 2023-05-20:
Hi, this is the first time that I see this error. This is not logic and I just checked the code to see if I was wrongly trying to open a pickle file in "write binary" mode. It is not the case, it is correctly read in "read binary" mode, so the code is ok. You must have run on a strange condition where the pickle file was not correctly saved?
Did you tried launching the optimization task to regenerate that file?
That file is used to survive HA restarts.

jahpster, 2023-05-22:
Hi,
I reinstalled the addon and reset the configuration file and have still the same error.
I tried launching the optimization task but I can't connect .
Are there files who persist after deleting the addon?

davidusb-geek, 2023-05-22:
Have you ever been able to launch an optimization task successfully with the add-on?
Otherwise, yes there are some permanent files in the `/share` folder. You can manually erase them to completely remove any previous results.

jahpster, 2023-05-23:
I deleted the files in the share folder and now everything is working fine.
I must have done something wrong with the configuration file.
Thanks for the help!

closed 2023-05-23 by jahpster