davidusb-geek/emhass #82: Feature Request: unit_prod_price & unit_load_cost forecasts

purcell-lab, 2023-05-28:
Thanks for implementing the price and cost sensors.

It would be great if this could be extended (like the other EMHASS published sensors) to include the future forecasts as an atribute.

[image]

[image]

[image]

davidusb-geek, 2023-05-29:
Hi, yes sure. This should be an easy fix. It can be added in the next release.

purcell-lab, 2023-06-05:
I don't seem to be getting much through these attributes ;-(

They are all coming out as 0.0.

[image]
[image]

Octofinger, 2023-06-25:
Same issue here. It seems the value is rounded to nearest whole number at some point and then converted to a float again. Here's what it looks like for me (sorry for the large screenshot but I wanted to the the 2.0 values included):
[image]

davidusb-geek, 2023-06-26:
This is an easy fix. It is already solved here: #91
It will be available on the next release.

davidusb-geek, 2023-06-29:
I've just published a new release, it should work as expected.

purcell-lab, 2023-06-30:
Unfortunately I am still doing the rounded to 0.0 on the forecasts so this issue should be reopened.

[image]

davidusb-geek, 2023-07-01:
Ok, it works as expected on the tests. I'll take a look at this

purcell-lab, 2023-07-01:
Let me know if you want me to test any code.

davidusb-geek, 2023-07-02:
Thanks Mark. Just a question this is happening for both unit_prod_price and unit_load_cost?

Octofinger, 2023-07-02:
I can concur. I have the same issue after updating to 0.4.13. Both `unit_prod_price` and `unit_load_cost` have attributes rounded to nearest integer, even if they are floats.
[image]

[image]

purcell-lab, 2023-07-02:
> Thanks Mark. Just a question this is happening for both unit_prod_price and unit_load_cost?

Yes both are being rounded to 0.0

purcell-lab, 2023-07-13:
This is amazing:
https://chat.openai.com/share/6a5bd9ed-f67f-4c32-a5d7-96e06de1255d

[image]

--- retrieve_hass.py
+++ retrieve_hass_modified.py
@@ -231,7 +231,7 @@
             datum[entity_id.split('sensor.')[1]] = vals_list[i]
             forecast_list.append(datum)
         data = {
-            "state": str(state),
+            "state": "{:.2f}".format(state),
             "attributes": {
                 "unit_of_measurement": unit_of_measurement,
                 "friendly_name": friendly_name,
@@ -307,7 +307,7 @@
                                                     friendly_name, "scheduled_forecast", state)
         else:
             data = {
-                "state": str(state),
+                "state": "{:.2f}".format(state),
                 "attributes": {
                     "unit_of_measurement": unit_of_measurement,
                     "friendly_name": friendly_name

--- retrieve_hass_modified.py
+++ retrieve_hass_modified_v2.py
@@ -223,7 +223,7 @@
         list_df = copy.deepcopy(data_df).loc[data_df.index[idx]:].reset_index()
         list_df.columns = ['timestamps', entity_id]
         ts_list = [str(i) for i in list_df['timestamps'].tolist()]
-        vals_list = [str(np.round(i)) for i in list_df[entity_id].tolist()]
+        vals_list = [str(np.round(i, 2)) for i in list_df[entity_id].tolist()]
         forecast_list = []
         for i, ts in enumerate(ts_list):
             datum = {}

davidusb-geek, 2023-07-13:
Some of the nice usages for chatgpt ;-)

PR #96 (2023-07-13): Fix rounding for price & cost forecasts
Closes #82
closed 2023-07-13 by davidusb-geek