davidusb-geek/emhass #85: Solar forcast is broken

bakerboy908, 2023-06-01:
**Describe the bug**
When calling EMHASS with solarforcast the system fails

**To Reproduce**
`curl -i -H "Content-Type: application/json" -X POST -d ''{"solar_forecast_kwp":5}'' http://localhost:5000/action/dayahead-optim`

**Screenshots**

```2023-06-01 13:58:00,995 - web_server - INFO - Setting up needed data
2023-06-01 13:58:01,021 - web_server - INFO - Retrieving weather forecast data using method = solar.forecast
2023-06-01 13:58:02,577 - web_server - ERROR - Exception on /action/dayahead-optim [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2190, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1486, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1484, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1469, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 174, in action_call
    input_data_dict = set_input_data_dict(config_path, str(data_path), costfun,
  File "/usr/local/lib/python3.9/dist-packages/emhass/command_line.py", line 89, in set_input_data_dict
    df_weather = fcst.get_weather_forecast(method=optim_conf['weather_forecast_method'])
  File "/usr/local/lib/python3.9/dist-packages/emhass/forecast.py", line 273, in get_weather_forecast
    data_dict = {'ts':list(data_raw['result']['watts'].keys()), 'yhat':list(data_raw['result']['watts'].values())}
TypeError: 'NoneType' object is not subscriptable
2023-06-01 13:58:50,660 - web_server - INFO - Setting up needed data
2023-06-01 13:58:50,687 - web_server - INFO - Retrieving weather forecast data using method = solar.forecast
2023-06-01 13:58:51,871 - web_server - ERROR - Exception on /action/dayahead-optim [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2190, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1486, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1484, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1469, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 174, in action_call
    input_data_dict = set_input_data_dict(config_path, str(data_path), costfun,
  File "/usr/local/lib/python3.9/dist-packages/emhass/command_line.py", line 89, in set_input_data_dict
    df_weather = fcst.get_weather_forecast(method=optim_conf['weather_forecast_method'])
  File "/usr/local/lib/python3.9/dist-packages/emhass/forecast.py", line 273, in get_weather_forecast
    data_dict = {'ts':list(data_raw['result']['watts'].keys()), 'yhat':list(data_raw['result']['watts'].values())}
TypeError: 'NoneType' object is not subscriptable
```

**Home Assistant installation type**
 - Home Assistant OS

**Your hardware**
- OS: HA OS
- Architecture: amd64

**EMHASS installation type**
 - Add-on

**Additional context**
Add any other context about the problem here.

davidusb-geek, 2023-06-01:
Did you tried with the following syntax?
```
curl -i -H "Content-Type:application/json" -X POST -d '{"solar_forecast_kwp":5}' http://localhost:5000/action/dayahead-optim
```

davidusb-geek, 2023-06-03:
This is working perfectly fine. I've just tested it right now with that syntax on my latest answer.

[image]

bakerboy908, 2023-06-11:
My appoligies, looks like im having problems with the shell scripts in home assistnat

closed 2023-06-11 by bakerboy908