davidusb-geek/emhass #90: Add configuration to prefix all published sensors

Octofinger, 2023-06-25:
**Describe the feature request**
I want to be able to configure a sensor prefix for all sensors published to HA from EMHASS.

**Expected behavior**
In config file or as parameters in the call to the ``publish-data`` endpoint, a prefix string can be configured that will prefix all sensors published to Homeassistant.

E.g. Configuring the string "emhass_" as prefix would automatically nameall sensors published to the form ``sensor.emhass_<sensor_name>``.

The reason for this request is that it will be much easier to find the emhass specific sensors in HA. It's also a bit more future proof as any added sensors in the future will also be automatically prefixed. This makes the need to update documentation, and ``publish-data`` endpoint, less critical, for those in need of renaming published sensors just to add prefixes. As an example to prove my point, we have gotten two new parameters added: ``unit_load_cost`` and ``unit_prod_price`` and they are both undocumented in the [https://emhass.readthedocs.io/en/latest/intro.html#the-publish-data-specificities](https://emhass.readthedocs.io/en/latest/intro.html#the-publish-data-specificities) section and, afaik, has no custom renaming features yet.

davidusb-geek, 2023-06-25:
All this is fully implemented.
But you are right the documentation is not up to the latest because support for custom names for `load_cost` and `prod_price` were added later.
Just pass your custom names as you would with the other sensors by using the keys `custom_unit_load_cost_id` and `custom_unit_prod_price_id`. I will try to update the documentation as soon as I can.

davidusb-geek, 2023-06-26:
Done I've just updated the documentation. 
Again, all the published sensors are fully configurable when calling the `publish-data` endpoint.
Closing this but feel free to reopen if you have any issue.

Octofinger, 2023-06-26:
I believe you missed my point. I'm aware that each sensor can be _individually_ renamed.

My suggestion is to add _one_ configuration that adds a prefix to _all_ sensors.
E.g.:
``sensor_prefix: 'emhass_'``

would make all original sensor names be prefixed like:
``sensor.emhass_p_batt_forecast``
``sensor.emhass_p_grid_forecast``
``sensor.emhass_p_load_forecast``
``sensor.emhass_unit_load_cost``
etc...

davidusb-geek, 2023-06-28:
I will take a look at this. It could be a good thing to add a prefix just not by default. It will be up to the user to activate that.

davidusb-geek, 2023-06-29:
I've just released a new version that includes this improvement.
You should be able to pass a `publish_prefix` key containing your desired prefix when calling the publish-data endpoint.

closed 2023-06-30 by davidusb-geek

Octofinger, 2023-07-02:
I have tested it and it works the charm. Thank you!

Octofinger, 2023-07-02:
Just noted an unexpected side effect of this setting. When setting a prefix, not only are sensor names prefixed, but also sensor attribute dict keys. E.g:
[image]

As long as this is documented, I guess I have no objection.

davidusb-geek, 2023-07-03:
Ok, it was added to the docs here: https://emhass.readthedocs.io/en/latest/intro.html#passing-other-data