davidusb-geek/emhass #93: When using the custom names there are no attributes is the sensors

gieljnssns, 2023-06-26:
**Describe the bug**
When I use custom entity names there are no attrfibutes exposed
**To Reproduce**
Steps to reproduce the behavior

**Expected behavior**
I expected to see the attributes

**Screenshots**
[image]

[image]

[image]

[image]

**Home Assistant installation type**
 - Home Assistant OS

**Your hardware**
- OS: HA OS
- Architecture: amd64

**EMHASS installation type**
 - Add-on

**Additional context**
Add any other context about the problem here.

davidusb-geek, 2023-06-26:
I see why this is happening. It is a bug. I will fix it as soon as possible.

davidusb-geek, 2023-06-29:
I've just published a new release solving this issue.

closed 2023-06-30 by davidusb-geek