davidusb-geek/emhass #94: Feature request - pass current battery SOC to dayahead-optim to improve optimization

Octofinger, 2023-06-28:
I find it that `dayahead-optim` quite often get my battery strategy wrong since the battery has a different SOC than `dayahead-optim` seems to assume. For different reasons, my battery level might not be what the optimization expects due to e.g. unexpected load, or unexpected PV power. As of now it always seems to assume that the battery has `SOCtarget` charge at the beginning of an optimization period.

Allow me to pass a parameter to `dayahead-optim` where I can send current battery SOC to help the optimization make the right decisions.

davidusb-geek, 2023-06-29:
I just treated this, you will be able to pass the SOCtarget during runtime. This will be available later today after the new release is published.

davidusb-geek, 2023-06-29:
I've just published the new release with this added improvement.

closed 2023-06-30 by davidusb-geek

Octofinger, 2023-07-02:
This didn't turn out as expected. Now I can pass `SOCtarget`, but unfortunately `dayahead-optim` uses this value _both as start value and target value_.

The issue I have is that I want to have a target SOC of 60%, but at the time I call `dayahead-optim`, the current battery SOC is, e.g., 90%. How can I tell the optimization that the battery already has 90% SOC when I start the optimization? Right now it assumes that the current battery SOC is the same as I want it to be in 24 hours, which is rarely the case.

Or did I just miss something here?

davidusb-geek, 2023-07-02:
Oh yes for the day-ahead optimization a hard condition is SOCinit = SOCfinal.
What you want to do can be achieved by just using the MPC call. You can set the prediction horizon to 24h for the MPC to achieve what you want.