davidusb-geek/emhass #100: Help request: Ml-forecaster gives “All arrays must be of the same length”

overas, 2023-07-27:
I need som help with using mlforecast. Emhass addon gives error about “All arrays must be of the same length” when I run the shell command for dayahead-optim when I use load forecast method “mlforecaster” as the load forecast metode. When I use load forecast method it work as expected.

I have posted my problem earlier in Home Assistant Forum:
https://community.home-assistant.io/t/emhass-an-energy-management-for-home-assistant/338126/1017?u=haraldov

I hope you can give me some help David.

overas, 2023-07-27:
Emhass version: 0.3.15
## System Information

version | core-2023.7.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.4
os_name | Linux
os_version | 6.1.34
arch | x86_64
timezone | Europe/Oslo
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4775
Installed Version | 1.32.1
Stage | running
Available Repositories | 1326
Downloaded Repositories | 10

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | October 7, 2023 at 02:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | false
remote_server | eu-central-1-6.ui.nabu.casa
certificate_status | ready
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Easee EV Charger</summary>

component_version | 0.9.54
-- | --
reach_easee_cloud | ok
connected2stream | true

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.3
-- | --
update_channel | stable
supervisor_version | supervisor-2023.07.1
agent_version | 1.5.1
docker_version | 23.0.6
disk_total | 116.7 GB
disk_used | 14.1 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.6.0), Advanced SSH & Web Terminal (15.0.5), ESPHome (2023.7.0), Z-Wave JS UI (1.15.2), EMHASS (0.3.15), SQLite Web (3.9.0), Mosquitto broker (6.2.1), Frigate (0.12.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 1
views | 10
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | June 4, 2023 at 07:49
-- | --
current_recorder_run | July 27, 2023 at 21:37
estimated_db_size | 1113.00 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

davidusb-geek, 2023-07-27:
Hi I will take a deeper look at this and try to reproduce the issue. It might be a bug because you are using a 60min time step. I will try to test under these conditions. I'll keep you updated here.

overas, 2023-07-27:
Here are the log and shell command when I do first "ml_forecast_fit":
```
shell_command:
 ml_forecast_fit: "curl -i -H \"Content-Type:application/json\" -X POST -d '{\"num_lags\": 24}' http://localhost:5000/action/forecast-model-fit"
```

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun emhass (no readiness notification)
s6-rc: info: service legacy-services successfully started
2023-07-27 23:45:37,652 - web_server - INFO - Launching the emhass webserver at: http://0.0.0.0:5000
2023-07-27 23:45:37,652 - web_server - INFO - Home Assistant data fetch will be performed using url: http://supervisor/core/api
2023-07-27 23:45:37,652 - web_server - INFO - The data path is: /share
2023-07-27 23:45:37,653 - web_server - INFO - Using core emhass version: 0.4.14
waitress   INFO  Serving on http://0.0.0.0:5000
2023-07-27 23:45:44,025 - web_server - INFO - Setting up needed data
2023-07-27 23:45:44,072 - web_server - INFO - Retrieve hass get data method initiated...
2023-07-27 23:45:50,839 - web_server - INFO -  >> Performing a machine learning forecast model fit...
2023-07-27 23:45:50,840 - web_server - INFO - Performing a forecast model fit for load_forecast
2023-07-27 23:45:50,849 - web_server - INFO - Training a KNeighborsRegressor model
2023-07-27 23:45:50,892 - web_server - INFO - Elapsed time for model fit: 0.0424647331237793
2023-07-27 23:45:51,019 - web_server - INFO - Prediction R2 score of fitted model on test data: 0.6076818704445501
```

and here are the log and shell command when I do "ml_forecast_tune" after running "ml_forecast_fit":

```
shell_command:
ml_forecast_tune: "curl -i -H \"Content-Type:application/json\" -X POST -d '{}' http://localhost:5000/action/forecast-model-tune"
```

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun emhass (no readiness notification)
s6-rc: info: service legacy-services successfully started
2023-07-27 23:45:37,652 - web_server - INFO - Launching the emhass webserver at: http://0.0.0.0:5000
2023-07-27 23:45:37,652 - web_server - INFO - Home Assistant data fetch will be performed using url: http://supervisor/core/api
2023-07-27 23:45:37,652 - web_server - INFO - The data path is: /share
2023-07-27 23:45:37,653 - web_server - INFO - Using core emhass version: 0.4.14
waitress   INFO  Serving on http://0.0.0.0:5000
2023-07-27 23:45:44,025 - web_server - INFO - Setting up needed data
2023-07-27 23:45:44,072 - web_server - INFO - Retrieve hass get data method initiated...
2023-07-27 23:45:50,839 - web_server - INFO -  >> Performing a machine learning forecast model fit...
2023-07-27 23:45:50,840 - web_server - INFO - Performing a forecast model fit for load_forecast
2023-07-27 23:45:50,849 - web_server - INFO - Training a KNeighborsRegressor model
2023-07-27 23:45:50,892 - web_server - INFO - Elapsed time for model fit: 0.0424647331237793
2023-07-27 23:45:51,019 - web_server - INFO - Prediction R2 score of fitted model on test data: 0.6076818704445501
2023-07-27 23:49:04,257 - web_server - INFO - Setting up needed data
2023-07-27 23:49:04,263 - web_server - INFO - Retrieve hass get data method initiated...
2023-07-27 23:49:10,787 - web_server - INFO -  >> Performing a machine learning forecast model tune...
2023-07-27 23:49:10,788 - web_server - INFO - Bayesian hyperparameter optimization with backtesting
Number of models compared: 70,
         10 bayesian search in each lag configuration.

lags grid:   0%|          | 0/7 [00:00<?, ?it/s]
lags grid:  14%|█▍        | 1/7 [00:02<00:13,  2.17s/it]
lags grid:  29%|██▊       | 2/7 [00:02<00:06,  1.26s/it]
lags grid:  43%|████▎     | 3/7 [00:03<00:03,  1.14it/s]
lags grid:  57%|█████▋    | 4/7 [00:03<00:02,  1.44it/s]
lags grid:  71%|███████▏  | 5/7 [00:04<00:01,  1.69it/s]
lags grid:  86%|████████▌ | 6/7 [00:04<00:00,  1.86it/s]
lags grid: 100%|██████████| 7/7 [00:04<00:00,  2.01it/s]
lags grid: 100%|██████████| 7/7 [00:04<00:00,  1.43it/s]
2023-07-27 23:49:15,711 - web_server - INFO - Elapsed time: 4.922260284423828
2023-07-27 23:49:15,770 - web_server - INFO - R2 score for optimized prediction in train period: 0.6605145262953049
2023-07-27 23:49:15,777 - web_server - INFO - R2 score for optimized prediction in test period: 0.18052539993964367
2023-07-27 23:49:15,777 - web_server - INFO - Number of optimal lags obtained: 12
```

davidusb-geek, 2023-07-28:
Ok when you launched the `tune` method then the system found an optimal value for the `num_lags` parameter which is now 12 in your case. So refit your model specifying `num_lags:12` and then retry launching the optimization. This is just a test, I'm not sure that it is the final solution to this issue.

overas, 2023-07-28:
Using "num_lags: 12" in shell_command for "ml_forecast_fit" and then run "ml_forecast_tune":
```
shell_command:
 ml_forecast_fit: "curl -i -H \"Content-Type:application/json\" -X POST -d '{\"num_lags\": 12}' http://localhost:5000/action/forecast-model-fit"
 ml_forecast_tune: "curl -i -H \"Content-Type:application/json\" -X POST -d '{}' http://localhost:5000/action/forecast-model-tune"
```

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun emhass (no readiness notification)
s6-rc: info: service legacy-services successfully started
2023-07-28 13:05:22,540 - web_server - INFO - Launching the emhass webserver at: http://0.0.0.0:5000
2023-07-28 13:05:22,540 - web_server - INFO - Home Assistant data fetch will be performed using url: http://supervisor/core/api
2023-07-28 13:05:22,540 - web_server - INFO - The data path is: /share
2023-07-28 13:05:22,541 - web_server - INFO - Using core emhass version: 0.4.14
waitress   INFO  Serving on http://0.0.0.0:5000
2023-07-28 13:05:36,858 - web_server - INFO - Setting up needed data
2023-07-28 13:05:36,889 - web_server - INFO - Retrieve hass get data method initiated...
2023-07-28 13:05:43,376 - web_server - INFO -  >> Performing a machine learning forecast model fit...
2023-07-28 13:05:43,377 - web_server - INFO - Performing a forecast model fit for load_forecast
2023-07-28 13:05:43,387 - web_server - INFO - Training a KNeighborsRegressor model
2023-07-28 13:05:43,429 - web_server - INFO - Elapsed time for model fit: 0.04137992858886719
2023-07-28 13:05:43,486 - web_server - INFO - Prediction R2 score of fitted model on test data: 0.22598061958694815
2023-07-28 13:05:59,790 - web_server - INFO - Setting up needed data
2023-07-28 13:05:59,795 - web_server - INFO - Retrieve hass get data method initiated...
2023-07-28 13:06:06,796 - web_server - INFO -  >> Performing a machine learning forecast model tune...
2023-07-28 13:06:06,797 - web_server - INFO - Bayesian hyperparameter optimization with backtesting
Number of models compared: 70,
         10 bayesian search in each lag configuration.

lags grid:   0%|          | 0/7 [00:00<?, ?it/s]
lags grid:  14%|█▍        | 1/7 [00:01<00:11,  1.85s/it]
lags grid:  29%|██▊       | 2/7 [00:02<00:05,  1.17s/it]
lags grid:  43%|████▎     | 3/7 [00:03<00:03,  1.13it/s]
lags grid:  57%|█████▋    | 4/7 [00:03<00:02,  1.26it/s]
lags grid:  71%|███████▏  | 5/7 [00:04<00:01,  1.30it/s]
lags grid:  86%|████████▌ | 6/7 [00:05<00:00,  1.42it/s]
lags grid: 100%|██████████| 7/7 [00:05<00:00,  1.46it/s]
lags grid: 100%|██████████| 7/7 [00:05<00:00,  1.23it/s]
2023-07-28 13:06:12,518 - web_server - INFO - Elapsed time: 5.72133731842041
2023-07-28 13:06:12,546 - web_server - INFO - R2 score for optimized prediction in train period: 0.5664681225834161
2023-07-28 13:06:12,550 - web_server - INFO - R2 score for optimized prediction in test period: -1.1179312477729821
2023-07-28 13:06:12,551 - web_server - INFO - Number of optimal lags obtained: 12
2023-07-28 13:06:34,354 - web_server - INFO - EMHASS server online, serving index.html...
```

Change in emhass addon config from "naive" to "mlforecaster" and run the shell_command for "dayahead-optim":

```
  trigger_nordpool_forecast: "curl -i -H \"Content-Type: application/json\" -X POST -d '{
    \"load_cost_forecast\":{{((state_attr('sensor.nordpool', 'raw_today') | map(attribute='value') | list  + state_attr('sensor.nordpool', 'raw_tomorrow') | map(attribute='value') | list))[now().hour:][:24] }},
    \"prod_price_forecast\":{{((state_attr('sensor.nordpool_uten_avgifter', 'raw_today') | map(attribute='value') | list  + state_attr('sensor.nordpool_uten_avgifter', 'raw_tomorrow') | map(attribute='value') | list))[now().hour:][:24]}},
    \"def_total_hours\":{{states('sensor.list_operating_hours_of_each_deferrable_load')}}
    }' http://localhost:5000/action/dayahead-optim"
```

Emhass log give:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
[... 19 lines cut ...]
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1486, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1484, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1469, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 174, in action_call
    input_data_dict = set_input_data_dict(config_path, str(data_path), costfun,
  File "/usr/local/lib/python3.9/dist-packages/emhass/command_line.py", line 91, in set_input_data_dict
    P_load_forecast = fcst.get_load_forecast(method=optim_conf['load_forecast_method'])
[... cut ...]
  File "/usr/local/lib/python3.9/dist-packages/pandas/core/internals/construction.py", line 502, in dict_to_mgr
    return arrays_to_mgr(arrays, columns, index, dtype=dtype, typ=typ, consolidate=copy)
  File "/usr/local/lib/python3.9/dist-packages/pandas/core/internals/construction.py", line 120, in arrays_to_mgr
    index = _extract_index(arrays)
  File "/usr/local/lib/python3.9/dist-packages/pandas/core/internals/construction.py", line 674, in _extract_index
    raise ValueError("All arrays must be of the same length")
ValueError: All arrays must be of the same length
```

overas, 2023-08-09:
Don't quite know what I've done, but now I don't get "All arrays must be of the same length" when I run the shell command for dayahead-optim when I use load forecast method "mlforecaster" as the load forecast method. I am going to close the help request now and open it up if the problem is starting over. 
Thanks for your help and for your great work with emhass.

I run these shell commands:
```
shell_command:
  ml_forecast_fit: "curl -i -H \"Content-Type:application/json\" -X POST -d '{\"days_to_retrieve\": 40, \"num_lags\": 24}' http://localhost:5000/action/forecast-model-fit"

  ml_forecast_tune: "curl -i -H \"Content-Type:application/json\" -X POST -d '{}' http://localhost:5000/action/forecast-model-tune"

  trigger_nordpool_forecast: "curl -i -H \"Content-Type: application/json\" -X POST -d '{
    \"load_cost_forecast\":{{((state_attr('sensor.nordpool', 'raw_today') | map(attribute='value') | list  + state_attr('sensor.nordpool', 'raw_tomorrow') | map(attribute='value') | list))[now().hour:][:24] }},
    \"prod_price_forecast\":{{((state_attr('sensor.nordpool_uten_avgifter', 'raw_today') | map(attribute='value') | list  + state_attr('sensor.nordpool_uten_avgifter', 'raw_tomorrow') | map(attribute='value') | list))[now().hour:][:24]}},
    \"def_total_hours\":{{states('sensor.list_operating_hours_of_each_deferrable_load')}}
    }' http://localhost:5000/action/dayahead-optim"
```

```
2023-08-09 14:42:53,732 - web_server - INFO - Setting up needed data
2023-08-09 14:42:53,737 - web_server - INFO - Retrieve hass get data method initiated...
2023-08-09 14:43:28,343 - web_server - INFO -  >> Performing a machine learning forecast model fit...
2023-08-09 14:43:28,345 - web_server - INFO - Performing a forecast model fit for load_forecast
2023-08-09 14:43:28,354 - web_server - INFO - Training a KNeighborsRegressor model
2023-08-09 14:43:28,367 - web_server - INFO - Elapsed time for model fit: 0.012708663940429688
2023-08-09 14:43:28,450 - web_server - INFO - Prediction R2 score of fitted model on test data: 0.11602009999995055
2023-08-09 14:43:56,288 - web_server - INFO - Setting up needed data
2023-08-09 14:43:56,290 - web_server - INFO - Retrieve hass get data method initiated...
2023-08-09 14:44:03,590 - web_server - INFO -  >> Performing a machine learning forecast model tune...
2023-08-09 14:44:03,591 - web_server - INFO - Bayesian hyperparameter optimization with backtesting
Number of models compared: 70,
         10 bayesian search in each lag configuration.

lags grid:   0%|          | 0/7 [00:00<?, ?it/s]
lags grid:  14%|█▍        | 1/7 [00:02<00:13,  2.30s/it]
lags grid:  29%|██▊       | 2/7 [00:02<00:06,  1.23s/it]
lags grid:  43%|████▎     | 3/7 [00:03<00:03,  1.11it/s]
lags grid:  57%|█████▋    | 4/7 [00:03<00:02,  1.34it/s]
lags grid:  71%|███████▏  | 5/7 [00:04<00:01,  1.50it/s]
lags grid:  86%|████████▌ | 6/7 [00:04<00:00,  1.62it/s]
lags grid: 100%|██████████| 7/7 [00:05<00:00,  1.71it/s]
lags grid: 100%|██████████| 7/7 [00:05<00:00,  1.30it/s]
2023-08-09 14:44:09,003 - web_server - INFO - Elapsed time: 5.412102937698364
2023-08-09 14:44:09,125 - web_server - INFO - R2 score for optimized prediction in train period: 0.663039229812043
2023-08-09 14:44:09,130 - web_server - INFO - R2 score for optimized prediction in test period: -1.3697067069349802
2023-08-09 14:44:09,130 - web_server - INFO - Number of optimal lags obtained: 48
2023-08-09 14:44:40,581 - web_server - INFO - Setting up needed data
2023-08-09 14:44:40,584 - web_server - INFO - Retrieving weather forecast data using method = scrapper
2023-08-09 14:44:42,609 - web_server - INFO - Retrieving data from hass for load forecast using method = mlforecaster
2023-08-09 14:44:42,610 - web_server - INFO - Retrieve hass get data method initiated...
2023-08-09 14:44:45,897 - web_server - INFO -  >> Performing dayahead optimization...
2023-08-09 14:44:45,897 - web_server - INFO - Performing day-ahead forecast optimization
2023-08-09 14:44:45,909 - web_server - INFO - Perform optimization for the day-ahead
2023-08-09 14:44:45,962 - web_server - INFO - Status: Infeasible
2023-08-09 14:44:45,962 - web_server - INFO - Total value of the Cost function = -113.54
```

closed 2023-08-09 by overas