davidusb-geek/emhass #101: Feature Request: Move config files to separate config folder

treynaer, 2023-07-31:
Hi,
I'm running EMHASS on unRAID, unfortunately docker on unRAID only supports path mapping, no individual files.
Currently I map the whole app folder. EMHASS works but I need to update the app folder manually after every update.
Do you see the option to move all config related files to a separate config folder?
This will allow proper path mapping on unRAID.
Regards,
Tim

treynaer, 2023-08-10:
Hi,
I indeed use the docker method. However the docker implementation in unRaid only allows me to map a complete folder of the filesystem to a folder in the docker container. I cannot map an individual file as whats proposed in the documentation.
I need to map a complete folder to the docker container that should contain the config files as well as all the files needed to run EMHASS. This is because EMHASS expects the config.yaml and secrets.yaml be in the root of EMHASS.

If EMHASS is adapted so that the config and secrets file in a config folder as how you manage the data folder I can map only that container folder to a folder on the filesystem. 
EMHASS scripts stay within the container, "user" files can be mapped outside the container by mapping the /config and /data folders to the filesystem of the docker host.
Regards,
Tim

davidusb-geek, 2023-08-18:
Ok, I get it now. I'll take a look at this and possibly add an additional flag to be used to provide a folder where you can put your configuration files.

davidusb-geek, 2023-08-20:
Ok so the code is just expecting that `config_emhass.yaml` and `secrets_emhass.yaml` are located in the `app/` folder inside the docker container.

So first solution will be to set the volume to map your local folder. 
Something like this:
```
docker run -it --restart always -p 5000:5000 -e "LOCAL_COSTFUN=profit" -v $(pwd):/app --name DockerEMHASS <REPOSITORY:TAG>
```
But I'm not sure if this will copy the configuration files inside your local folder.

Otherwise a second solution will to mount and copy the files like this:
```
docker run -it --restart always -p 5000:5000 -e "LOCAL_COSTFUN=profit" --mount type=bind, source="$(pwd)"/emhass_config_files, target=/app cp <files> /app --name DockerEMHASS <REPOSITORY:TAG>
```
You will need to change the `cp` part to test if it works for copying your files.

treynaer, 2023-09-17:
Hi,
This is how I map the config and secrets file now.
The only issue with that is that I now have to maintain the whole app folder outside the docker container because its mapped to a local folder. This means that I manually have to copy the git repo to this local folder to update EMHASS as well as updating the container itself. This beats the purpose of using a docker container.

If config and secrets would move to for example /app/config I would be able to only map /app/config to a local folder.
The files that makeup EMHASS are untouched since they stay part of the docker container.

Regards,
Tim

davidusb-geek, 2024-01-29:
With the new version you can map the configuration an secrets file using the `CONFIG_PATH` and `SECRETS_PATH` environment variables respectively when using the `docker run` command.

I will be closing this as completed now but feel free to re-open if needed.

closed 2024-01-29 by davidusb-geek