davidusb-geek/emhass #104: Feature Request: Numerical Syntax

JoshC1994, 2023-08-16:
**Describe the bug**
The Syntax of numbers shown in the "Last Run Optimisation Results Table" sometimes displaying as scientific numbers.
[image]

**Why**
The fact may be simply due to the input data EMHASS is being fed and the maths being calculated so that the resulting number is too great for the plotly table to display.

**Fix**
Would it cause too many issues with resolution if EMHASS only considered data and calculated the resultant to say 2 decimal points. I would think that any more resolution is only adding complexity and would require unnecessary compute resources.
Maybe the decimal count could be set via config?       

I might be missing something but in most use cases I think that 2 decimal places would be enough resolution.

purcell-lab, 2023-08-16:
Thanks @JoshC1994 , I have been having this issue as well.

@davidusb-geek while you are fixing the rounding could I suggest some additional improvements:

Could I suggest that all the power values be rounded to 0 decimal places.

P_PV and P_Load are already displayed as int, but the other P_ variables are displayed with a lot of unnecessary precision and it would be good if they could also be displayed as int, cost_profit & cost_fun could also potentially be rounded to 2 decimal places:

[image]

davidusb-geek, 2023-08-18:
Solved here: #105
Will be available in the next release.

kcoffau, 2023-08-22:
+1 thank you! @davidusb-geek 

Appreciate everything you do for this project!

purcell-lab, 2023-09-03:
Nice work, much more readable:

[image]

davidusb-geek, 2023-09-06:
Great! Closing this as completed.

closed 2023-09-06 by davidusb-geek

purcell-lab, 2024-01-08:
I know this issue is closed, but the Power values seem to have snuck back in with unnecessary decimal (.0) again.
[image]

davidusb-geek, 2024-01-28:
This is solved in the (just-to-be-released) new release.