davidusb-geek/emhass #107: Q: Load Battery from grid?

sysrun, 2023-08-29:
Hi,

i am looking into this project for my home.

- 10.2kWp PV
- 9,6kWh Battery
- Dynamic electricity price (tibber).

My question: is it possible to configure emhass in a way that the battery can be filled from the grid IF the sun forecast is predicting bad performance from the pv AND the price is low.

Idea is to use the Battery in the winter as the price over a day can have differences up to 40 Cent

davidusb-geek, 2023-08-30:
EHMASS will provide you with the best possible schedule. No need to configure anything to obtain the kind of behavior that you describe. Charging from the grid for bad PV performance ahead and lower prices will be a natural solution that the algorithm will find, it is a very common scenario. But there is no need to provide additional configuration, that will just be the result of the optimization. I don't know at what stage you are right now but what you can do is install the add-on, provide correct data to it and analyze the daily results that it will obtain. Just analyze the results, no need to use the results to control or manage your actual loads as a first approach. Then when you feel confidence on these results where you should normally obtain the behavior that you described, then you can use the optimized schedule from EMHASS to control your loads automatically.

clunit, 2023-09-14:
Hi, 
I have a related question to battery charge/discharge. 
Is it possible to add a 'cost' to battery usage? For example let's say it's not economically feasible to charge/dischare unless the price difference is higher than a certain value. This to consider capital cost and battery lifespan. Or is this available somewhere that I haven't found?
BR,
Carl

davidusb-geek, 2023-09-26:
> Hi, I have a related question to battery charge/discharge. Is it possible to add a 'cost' to battery usage? For example let's say it's not economically feasible to charge/dischare unless the price [...]

This is related to #111 
Look at my answer there.

rwunsch, 2023-11-08:
I will be attempting exactly the same - load from GRID in winter using Tibber.
I managed to load on the lowest price of the day using NodeRed , which I programmed yesterday.
Getting more "logic" into this brought me to EMHASS ... 
I still have no idea how this works - but I am looking forward to explore!

closed 2023-12-19 by davidusb-geek