davidusb-geek/emhass #109: Feature request: Add a published parameter with the result of latest optimization

Octofinger, 2023-09-02:
I'd like to request an additional value that is passed back as a published value. The value shows the status of the latest performed optimization.

If I trigger an optimization, I can't tell from currently published values, whether the optimization ended up "Optimal" or "Infeasible", besides detecting abnormal suggestions in the optimization, like negative SOC or similar.
If the latest optimization status could be published, I could have HA detect this and avoid acting on the optimization results, and instead send me a notification that something may be wrong with my setup.

Having a sensor like ``latest_optim_status`` taking the values "Optimal" or "Infeasible", or any other value an optimization status might have, would help me a lot.

davidusb-geek, 2023-09-03:
Hi. Yes I understand that this can be helpful.
I just began to work on this but I just realized that this will need some more time because there is a number of modifications to be made. I will release a new version that will not contain this but I will try to work on this for the next release.

davidusb-geek, 2023-12-17:
Avaliable in the new version

closed 2023-12-17 by davidusb-geek