davidusb-geek/emhass #111: Feature request, use cost of ware on battery in optimization

lovekull76, 2023-09-15:
### Feature request
Use cost of ware on battery in optimization.

### Background
My battery, like everyone else's, degrade over time when used. There is a theoretical constant cost/cycle and/or maybe per kWh charged/discharged we could use.

The price fluctuations over 24h is not that great here in Sweden right now (and has not been for a long time), BUT:
Import price/kWh = spot price + transfer cost + constant energy tax + VAT (about spot price + 0.45 SEK, depending on spot price*VAT).
Export price/kWh = spot price + 0.6 SEK tax return from the government.
**So at any given hour the export price is always about 0.15 SEK higher than import.**

I use the actual hourly prices (import and export) in EMHASS and I think EMHASS goes crazy on the fact that export is always higher than import.
Below is a typical day here in Sweden, spot price fluctuation (max - min) over 24h is about 0.07 SEK (about 0.007 EUR) and EMHASS thinks I should cycle my battery about 7 times during these 24h!! The cost of these cycles is in the order of 70 SEK (see below)!

[image]
[image]

It would be nice if EMHASS could take the cost of cycling the battery into account when doing the optimization. 
I guess I could subtract the cost of ware/kWh from the export hourly price and feed it into EMHASS and see what happens, but then solar energy produced would also be influenced by this lower export price and mess up the optimization. And the ware is also present on battery **charge** and that makes it even more complex.

### Thoughts on battery ware (don't read if you don't want to sleep badly because of your investment)
The purchase price (excluding install) for my 10kWh battery was about 120k SEK inc. VAT.
I got about 50% in return from the government on the battery install, not really true due to a limit in what you can get from the government and my total install cost for PV and battery was way beyond that limit, but for the argument sake lets say I paid 60K SEK for it.
The life time of the battery is said to be 6000 cycles, that is 10 SEK per cycle!!! For my 10kWh battery that would mean the difference in import and export price would need to be about 1 SEK to be profitable, that will not happen very often so I guess the battery install was actually a very bad idea 🤯 But it feels right anyway 😄

davidusb-geek, 2023-09-15:
It could be nice to add a new special and direct constraint on the battery to limit its cycling and hence helping with ageing. 
I'll try to work on this in the near future.
For now what you can do is to use a dynamic imitation for the battery power. You will need to tune those limits values. From what I can see on your graphs a maximum battery power dynamic of 1 to 2 kW per observation period may be a good starting point. The same goes with the minimum battery power dynamic, you can try -1 to -2 kW and see what results you obtain. This will directly lead to less cycling at some extent. But as it is not very intuitive the best will to add a direct penalty on the cost function provided by the user.

nicolabortignon, 2023-10-16:
Hi @davidusb-geek, great work on EMHASS.
Out of curiosity, how would you model the penalty in the linear programming solution with pulp? 

There are mainly 5 factors in battery economics: 
1- how many cycles they can go through in their lifetime
2- how long is the lifetime of the chemical compound (aka, after how many years the battery will stop working independently of the utilization)
3- initial of the battery
4- cost of money (basically how much is worth the cash spent in t0, at any given point in time of the battery life). 
5- battery degradation (usually express in a yearly %, on total capacity). 

I'm asking as I'm trying to write a LP to compute the maximum ROI of a battery arbitrage system based on the last year of energy grid (buy/sell) prices.

enoch85, 2023-11-02:
@lovekull76 Sorry for a bit of OT, but I'm trying to automate this myself to decide when to charge an discharge.

> Import price/kWh = spot price + transfer cost + constant energy tax + VAT (about spot price + 0.45 SEK, depending on spot price*VAT).
Export price/kWh = spot price + 0.6 SEK tax return from the government.

Maybe I'm a bit tired now (I am), but but lets assume this:
- Import: spot price (50 öre) + transfer fees (in my case 59.40 öre - enery tax included) + VAT (25%) = **136.75 öre**
- Export:  Spot price (50 öre) + 14 transfer fee (back to me) + 60 öre from government = **124 öre**

> So at any given hour the export price is always about 0.15 SEK higher than import.

So what you're saying is that I will always earn more by exporting, than importing? Have I missed something?

To avoid cycling the battery when it's not worth it, I try to automatically calculate the prices by taking the spot price at the time my cars are smart charging (it's the cheapest), then calculate the difference in a function of different levels where I want to discharge the battery or not (BYD 11 kWh). In the function I take the "Import price" as per above, minus the actual spot price when charging, but reading this now I'm not 100% sure that's correct?

Again, sorry for OT - but I plan to use EMHASS soon, and doing some digging first. Thanks!

[image]
[image]

Function:
```
var current_spotpris = flow.get("cheap_spotpris");

// Skånska energi avgifter + moms
var max_spotpris = (current_spotpris + 59.40) * 1.25;

// create a new variable to hold with the max price included
flow.set("break_even", max_spotpris);

return [{ payload: flow.get("break_even") }]
```

davidusb-geek, 2024-01-04:
> Hi @davidusb-geek, great work on EMHASS.
Out of curiosity, how would you model the penalty in the linear programming solution with pulp?

I have just released a new version that implement a solution for this issue, adding some additional weight to the main cost function. 
To partially answer your question I will quote myself from the HA forum. 
These weights have the same units as the energy cost, so “currency”/kWh.
A value of 1 will surely have an impact on your battery schedule. These can be used to directly penalize battery usage and account for battery degradation. Set it to zero to avoid using them if you have a battery. Maybe it could be a good idea to set it to zero by default, I may change that in a future release.
This value could be set to a real value that could be computed using the CAPEX cost of the battery and knowing the degradation rate per kWh output. Unfortunately this information is not easy to come by and it will depend on the specific battery chemistry, characteristics and some of the factors that you cited.
So we should be able to lump the factors you cited into this single value but getting a real meaningful value will be hard.

davidusb-geek, 2024-01-04:
Closing this for now but reopen if needed.

closed 2024-01-04 by davidusb-geek