davidusb-geek/emhass #112: Controlling a heat pump

tmp-hallenser, 2023-09-15:
Summary: How to include a heat pump into the optimization?

Details:
* A heat pump is used for hot water and for the normal heating
* Both can only be controlled to a certain extend, i.e. you can trigger the heat pump by raising the temperature threshold 
* As soon as the temperature falls below a certain threshold, the heat pump kicks in

Hence, the parameters to controls are (both, for hot water and for normal heating)
* time of increasing the temp threshold to use cheap energy
* temperature delta of the increase

davidusb-geek, 2023-09-16:
Hi,

This is a known limitation of EMHASS. The best option will be to model the heat pump equations and include it in the optimization. It is in the roadmap for future works/improvements.

However, as it is today you can try a solution. This will be to find a simple model that relies that temperature delta with the power consumption of your heat pump. If you have history values recorded then it may be fairly easy to find a linear equation like this `power_heat_pump = constant * delta_temperature`. If you are able to fit a good value for that constant then just take the power schedule from EMHASS and compute the required temperature delta to send to your heat pump. If a linear equation dos not fit well then you can try a quadratic equation. If you have historical values I'm willing to help you on finding a good model for this.

Another option is to ask in the community forum for other people facing the same problerm and what are their implemented solutions.

purcell-lab, 2023-09-17:
A thermal model inside EMHASS would be great, in the meantime I'm setting def_hours to be responsive to external temperatures or costs to set the run times for my heat pumps.

I have previously had two discussions regarding scheduling of heat pumps:

Hot Water system: 
https://community.home-assistant.io/t/emhass-an-energy-management-for-home-assistant/338126/175?u=markpurcell

HVAC:
https://community.home-assistant.io/t/emhass-an-energy-management-for-home-assistant/338126/286?u=markpurcell

Here are the rules I'm using to set def_hours:

```yaml
    - name: def_total_hours_hvac
      unique_id: 466c12fd-269b-4292-8e50-d9f3f40dd7fa
      state: "{{is_state('automation.p_deferable3_automation','on')|int
               * state_attr('weather.noosa_heads_hourly','forecast')[:24]|selectattr('temperature','ge',states('input_number.emhass_deferrable3_set_point')|int)|map(attribute='temperature')|list|count}}"

    - name: def_total_hours_hws
      unique_id: ffff343f-d08d-4cbf-b783-3c8c116f0fd3
      state: "{{is_state('automation.p_deferrable_4','on')|int
               * max(6,((state_attr('sensor.amber_feed_in_forecast', 'forecasts')|selectattr('descriptor','eq','extremely_low')|list|count)/2))|int(0)}}"

    - name: def_total_hours_pool_heatpump
      unique_id: ad586a0f-d16a-4ab9-841d-96b753612709
      state: "{{is_state('automation.p_deferable1_automation','on')|int
               * max(0,((state_attr('sensor.amber_feed_in_forecast', 'forecasts')|selectattr('descriptor','eq','extremely_low')|list|count)/2)
                        )}}"
```

The results:
[image]

[image]

sladey, 2023-09-29:
I am just bringing a heat pump online with my swimming pool.  I am not (yet) on Amber as my retailer so I cannot use the approach of running the heat pump for as many "low price" hours as the Amber price forecast indicates.  My heatpump can product between 6-24kW of heating output at an input of .5-4.8kW.    According to the table here https://madimack.com/au/blog/pool-heating-how-to-calculate-heat-pump-sizes I can get a temperature increase for my pool (55K liters) in degrees per hour at a certain power output.  

So if I take the delta between setpoint and current temp (eg 2 degrees) then if I assume I need maximum power output (eg 24kw) then I need maximum power input (eg 4.8kW) and I need to set def_hours to "temperature delta/0.33".   the 0.33 comes from that table when the pool volume is approx 55m cubed and the power output is 24kW.   

Does that sound like it will work?

marschallphilipp, 2023-11-29:
@davidusb-geek thanks for this great project!
I'm experiencing one more constraint for planning the optimal time slots of our heart pump.
Since we own a machine which drives the heating and makes hot water too it is difficult to get the best results.
Reason is, that I would like to use two separate deferrable loads but these can not operate at the same time, as it's the same machine.
With just one deferrable load (e.g. 9 hrs) it would work just fine but I'm sure to get better results with two separate ones (e.g. 6 + 3 hrs).
Anyone facing the same issue or can help with a workaround or idea how to handle this?

purcell-lab, 2024-03-18:
> a linear equation like this `power_heat_pump = constant * delta_temperature`. If you are able to fit a good value for that constant then just take the power schedule from EMHASS and compute the required temperature delta to send to your heat pump.

I am having very good results with a constant of 1500 - 2000 W/°C, each house is different so please conduct your own measurements.
https://community.home-assistant.io/t/optimizing-hvac-air-conditioning-efficiency-with-surplus-solar-energy/658466?u=markpurcell

The adaptation of this is then straight forward into EMHASS:
```yaml
    - name: def_total_hours_hvac
      unique_id: 466c12fd-269b-4292-8e50-d9f3f40dd7fa
      state: "{{(is_state('automation.p_deferable3_automation','on')|int(0) + is_state('automation.p_deferrable3_hvac_v2','on')|int(0))
               * state_attr('weather.noosa_heads_hourly','forecast')[:(states('sensor.prediction_horizon')|int(0)/2)|int(0)]|
               selectattr('temperature','ge',states('input_number.emhass_deferrable3_set_point')|int)|map(attribute='temperature')|list|count}}"
```
[image]

[image]

[image]

maakuth, 2024-03-21:
Somebody has done [an LP solution](https://community.home-assistant.io/t/optimizing-heat-pump-w-accumulator-tank-run-times-to-heat-home-and-produce-hot-water-for-lowest-cost-using-nordpool-hourly-price/637562) for their heat pump control, but as of yet, they've not opened their source code.

borisvettorato, 2024-05-05:
It would be nice to have emhass mimicing the homely thermostat, especially for air-source heat pumps. In this case you should also model the house as a battery with a minimum and maximum charge state and losing and gaining energy depending on the weather, mostly temperature and sun. This would both work for heating and cooling. Or maybee get the most out of HA and even integrate controlling of blinds and the ventilation bypass for free cooling!

davidusb-geek, 2024-07-13:
Please take a look at the new thermal model just recently released: [https://emhass.readthedocs.io/en/latest/thermal_model.html](https://emhass.readthedocs.io/en/latest/thermal_model.html)

This can be very well be used to control a heat pump or any type of heating device.
Closing this as completed for now.
Open a new issue if having trouble implementing that solution.

closed 2024-07-13 by davidusb-geek