davidusb-geek/emhass #116: not a valid add-on repository

maxxnet2, 2023-09-26:
**Describe the bug**
https://github.com/davidusb-geek/emhass is not a valid add-on repository

**To Reproduce**
Add the repository to HA

**Expected behavior**
Add the repository to HA

**Screenshots**
[image]
[image]

**Home Assistant installation type**
 - Home Assistant OS

**Your hardware**
- OS: HA OS
- Architecture: amd64

**EMHASS installation type**
 - Add-on

**Additional context**
Add any other context about the problem here.

gieljnssns, 2023-09-26:
I think you have to use this one. 
https://github.com/davidusb-geek/emhass-add-on

davidusb-geek, 2023-09-26:
> I think you have to use this one. https://github.com/davidusb-geek/emhass-add-on

Exactly, you are not using the correct URL for the add-on repository.
Closing as that will solve the issue.