davidusb-geek/emhass #122: Error running forecast-model-tune after updating to 0.4.2

vdiogo, 2023-10-20:
**Describe the bug**
After updating EMHASS to 0.4.2 as I was running the forecast-model-tune, I got an error (see print screen bellow).
I rolled back to 0.4.0 and no issues

**To Reproduce**
1. Update from 0.4.0 to 0.4.2 (OK)
2. Run forecast-model-fit (OK)
3. Run forecast-model-tune (NOK)

**Expected behavior**
Would expect same behavior as with previous version

**Screenshots**
[image]

**Home Assistant installation type**
 - Home Assistant OS

**Your hardware**
- OS: Linux
- Architecture: aarch64

**EMHASS installation type**
 - Add-on

**Additional context**
N/A

davidusb-geek, 2023-10-21:
If you run the fit first and then the tune (in that order) it doesn't works?

vdiogo, 2023-10-21:
> If you run the fit first and then the tune (in that order) it doesn't works?

Correct:

```
Run forecast-model-fit (OK)
Run forecast-model-tune (NOK)
```

davidusb-geek, 2023-11-01:
This is still not working?
Erase the pkl file in the share folder then retrain your model.

davidusb-geek, 2024-01-04:
This should be solved by erasing old pkl files in the shared folder. Closing this for now but reopen if needed.

closed 2024-01-04 by davidusb-geek