davidusb-geek/emhass #125: If too few days are available from HA recorder, data fetch fails

HACS-bank, 2023-10-26:
**Describe the bug**
If HA cannot provide as many days of history as requested, an error is logged.

**To Reproduce**
Set up a new recorder database - e.g. change to a different database.  The database now contains only data from today, not from yesterday.
Set up EMHASS.
Config UI does not allow "days to retrieve" to be set to 1 - the minimum is 2

**Expected behavior**
Data fetch should succeed (to allow a new user to continue to next steps) even though the requested days are not available.

**Screenshots**
```
2023-10-26 15:09:36,432 - web_server - INFO - EMHASS server online, serving index.html...
2023-10-26 15:09:36,437 - web_server - WARNING - The data container dictionary is empty... Please launch an optimization task
2023-10-26 15:09:40,653 - web_server - INFO - Setting up needed data
2023-10-26 15:09:40,719 - web_server - INFO - Retrieving weather forecast data using method = scrapper
2023-10-26 15:09:43,767 - web_server - INFO - Retrieving data from hass for load forecast using method = naive
2023-10-26 15:09:43,768 - web_server - INFO - Retrieve hass get data method initiated...
2023-10-26 15:09:43,776 - web_server - ERROR - The retrieved JSON is empty, check that correct day or variable names are passed
2023-10-26 15:09:43,776 - web_server - ERROR - Either the names of the passed variables are not correct or days_to_retrieve is larger than the recorded history of your sensor (check your recorder settings)
2023-10-26 15:09:43,776 - web_server - ERROR - Exception on /action/dayahead-optim [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1455, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 869, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 867, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 852, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 179, in action_call
    input_data_dict = set_input_data_dict(config_path, str(data_path), costfun,
  File "/usr/local/lib/python3.9/dist-packages/emhass/command_line.py", line 91, in set_input_data_dict
    P_load_forecast = fcst.get_load_forecast(method=optim_conf['load_forecast_method'])
  File "/usr/local/lib/python3.9/dist-packages/emhass/forecast.py", line 585, in get_load_forecast
    rh.get_data(days_list, var_list)
  File "/usr/local/lib/python3.9/dist-packages/emhass/retrieve_hass.py", line 147, in get_data
    self.df_final = pd.concat([self.df_final, df_day], axis=0)
UnboundLocalError: local variable 'df_day' referenced before assignment
```

**Home Assistant installation type**
 - Home Assistant OS

**Your hardware**
- OS: HA OS,
- Architecture: i586

**EMHASS installation type**
 - Add-on

**Additional context**
Add any other context about the problem here.

davidusb-geek, 2023-11-01:
Well this is just a requirement, we need at least two days of recorded data to produce a meaningful load power forecast.
What is the use of a fake successful data fetch if we cannot do anything after that because the lack of data?

closed 2023-11-26 by HACS-bank

tomasbedrich, 2024-02-05:
Just my two cents on this – while the requirement is perfectly reasonable, it would be nice to catch the situation and raise more meaningful error message.