davidusb-geek/emhass #126: Timezone issue due Winter/Summer time switch

rickroetenberg, 2023-10-28:
version: 0.4.2 EMHASS Add-on in HA

2023-10-28 10:24:04,479 - web_server - ERROR - Exception on /action/publish-data [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1455, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 869, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 867, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 852, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 179, in action_call
    input_data_dict = set_input_data_dict(config_path, str(data_path), costfun,
  File "/usr/local/lib/python3.9/dist-packages/emhass/command_line.py", line 64, in set_input_data_dict
    fcst = forecast(retrieve_hass_conf, optim_conf, plant_conf,
  File "/usr/local/lib/python3.9/dist-packages/emhass/forecast.py", line 160, in __init__
    self.forecast_dates = pd.date_range(start=self.start_forecast,
  File "/usr/local/lib/python3.9/dist-packages/pandas/core/indexes/extension.py", line 95, in method
    result = attr(self._data, *args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/pandas/core/arrays/datetimelike.py", line 1775, in round
    return self._round(freq, RoundTo.NEAREST_HALF_EVEN, ambiguous, nonexistent)
  File "/usr/local/lib/python3.9/dist-packages/pandas/core/arrays/datetimelike.py", line 1761, in _round
    return result.tz_localize(
  File "/usr/local/lib/python3.9/dist-packages/pandas/core/arrays/_mixins.py", line 84, in method
    return meth(self, *args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/pandas/core/arrays/datetimes.py", line 1043, in tz_localize
    new_dates = tzconversion.tz_localize_to_utc(
  File "pandas/_libs/tslibs/tzconversion.pyx", line 284, in pandas._libs.tslibs.tzconversion.tz_localize_to_utc
pytz.exceptions.AmbiguousTimeError: Cannot infer dst time from 2023-10-29 02:00:00, try using the 'ambiguous' argument

Octofinger, 2023-10-28:
Beat me to it. I run emhass docker and the ``naive-mpc-optim`` chokes when trying to optimize past the change to winter time (from cest to cet).

purcell-lab, 2023-10-28:
Our National Energy Market (NEM) doesn't change timezones, even though parts of the country do observe daylight savings time changes. It does make for confusing billing.

One suggestion which might work with EMHASS is to store time based data internally in a constant timezone for calculations, like UTC, and then convert to local time for external display/ consumption.

sti0, 2023-10-29:
One issue more: when using the Maschine learning forecaster (e.g predict). The next timeslot its forecasting (while writing) is 7:30 but should be 8:30 in Germany after the change of the DST. 

I don't know how the timezone is handled when using the HA Addon but might would be a good idea to allow setting the Linux TZ env through the Addon config and use this value within the programm itself.

sti0, 2023-10-29:
As this might be a different issue I filled another bug ticket.
https://github.com/davidusb-geek/emhass/issues/128

davidusb-geek, 2023-11-01:
All computations in EMHASS are made with data transformed to UTC. But the data fetch from HA is tz-aware, so when transformed to UTC holes appears in the timestamps.
I'm a bit clueless here. Maybe I will just try to catch the error and rebuild the timestamps and the DF for this specific scenario?

davidusb-geek, 2023-12-17:
Possible solved with new version.

davidusb-geek, 2024-01-04:
Closing this for now but don't hesitate to reopen if needed.

closed 2024-01-04 by davidusb-geek