davidusb-geek/emhass #128: Timezone not respected when using mlforecaster

sti0, 2023-10-29:
**Describe the bug**
After changing DST in Europe (CEST->CST) the output of the `mlforecaster` is in the wrong timezone (UTC instead local TZ).

**To Reproduce**
Predict a ML model

**Expected behavior**
Predict values in the right timezone

**Screenshots**
[image]

**Home Assistant installation type**
 - Home Assistant OS

**Your hardware**
- OS: HA OS
- Architecture: amd64

**EMHASS installation type**
 - Add-on

**Additional context**
Works using MPC:
[image]

ronaldt80, 2023-10-29:
Same here, i use mpc but all optimizations are one hour off.

adammcdonagh, 2023-10-29:
I'm going to assume my problem is related too? It started from yesterday morning... If I turn `mlforecaster` off and use just `naive` I get no errors, but obviously not great...

```
2023-10-29 21:09:52,111 - web_server - ERROR - Exception on /action/naive-mpc-optim [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1455, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 869, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 867, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 852, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 179, in action_call
    input_data_dict = set_input_data_dict(config_path, str(data_path), costfun,
  File "/usr/local/lib/python3.9/dist-packages/emhass/command_line.py", line 120, in set_input_data_dict
    P_load_forecast = fcst.get_load_forecast(method=optim_conf['load_forecast_method'], set_mix_forecast=True, df_now=df_input_data)
  File "/usr/local/lib/python3.9/dist-packages/emhass/forecast.py", line 615, in get_load_forecast
    forecast_out = mlf.predict(data_last_window)
  File "/usr/local/lib/python3.9/dist-packages/emhass/machine_learning_forecaster.py", line 206, in predict
    predictions = self.forecaster.predict(steps=self.lags_opt,
  File "/usr/local/lib/python3.9/dist-packages/skforecast/ForecasterAutoreg/ForecasterAutoreg.py", line 696, in predict
    if self.differentiation is not None:
AttributeError: 'ForecasterAutoreg' object has no attribute 'differentiation'
```

robinostlund, 2023-10-31:
If you are using docker, have you mounted this volume? - '/etc/localtime:/etc/localtime:ro'

adammcdonagh, 2023-10-31:
> If you are using docker, have you mounted this volume? - '/etc/localtime:/etc/localtime:ro'

I’m using the HA Addon. So that’s not configurable.

davidusb-geek, 2023-11-01:
This is all related to #126.
Catching this specific case should solve the issue. I need to spend some time on this.
However if you retrain the model the problem is still there?

sti0, 2023-11-01:
No, even after running fit, tune and predict ( in that order) the predict output is 1h behind in my case and prints UTC instead of the local time in the web ui (2023-11-01 08:00:00+00:00 instead of 9am). 

Just to mention again, MPC shows the right time values in the web ui. So I'm not sure if this is the same issue.

sti0, 2023-11-01:
BTW. The published sensor in HA shows the wrong timezone setting, too.

Compared with a deferrable load sensor:
[image]

[image]

davidusb-geek, 2024-01-04:
Hi is this still a current issue. There were some changes in time zone handling of the latest versions. Hopefully this was solved.

davidusb-geek, 2024-02-09:
I am closing this as the latest changes on handling TZ should improve this.
Also a solution could be to just retrain your ML forecaster model after the DST change.
Go to the `share` folder, erase the corresponding `.pkl` file, then retrain the ML forecaster model.

closed 2024-02-09 by davidusb-geek