davidusb-geek/emhass #129: Feature Request: Add possibility to add minimum power in P_deferrable_nom

robinostlund, 2023-10-31:
HI,
First of, awesome work with this tool it is awesome and looking forward for new features in upcoming releases 😄 

Anyway, i have a feature request and that is to be able to specify the minimal power that a defferable load can handle. For example i have an ev charger where i specify the max power it can use today but it can only charge my car if the power is above 6a per phase, so around 4140w is the minimum otherwise the charger wont start. 

Brg
Robin

davidusb-geek, 2023-11-01:
Yes this could be nice.
I've reluctant to add this because this will add more constraints to the optimization problem but I understand that it can be useful in some context.

davidusb-geek, 2024-02-04:
This was added to the development to do list.

Just for reference here is an example on how to implement the minimum power that a deferrable load can handle, otherwise it disconnect: self.optim_conf['P_deferrable_min'][k]. Use this: https://stackoverflow.com/questions/24449959/linear-programming-constraint-x-c-or-x-0

RudolfRendier, 2024-11-17:
Would this be an option for the battery as well?

I have a fixed rated charger, but I guess all batteries have a minimum C-rate.

davidusb-geek, 2024-11-17:
I've never heard such thing as a minimum C-rate. You'll have maximum charge and discharge C-rates but no minimum limitations

tbrasser, 2025-03-09:
> I've never heard such thing as a minimum C-rate. You'll have maximum charge and discharge C-rates but no minimum limitations

For LFP batteries low C charging can shorten its life. However when you have solar there's not much that can be done (except dual sol/uti charging to total some minimum C rate).

I have no solar yet, so having minimum batt charge power would be welcome!

davidusb-geek, 2025-08-26:
Implemented and merged this on PR #575.

Added some unit tests for this and passing fine.

Pass in the new parameters like this:

[image]