davidusb-geek/emhass #130: BUG: def_total_hours: 24 and set_def_constant: 1 cause tons on power on 1 hour

robinostlund, 2023-10-31:
HI,
Just noticed that when i set def_total_hours to 24h and set_def_constant to 1 makes the power increase to a very high value that is above my defined P_deferrable_nom. So it seems like a bug when specifying to run 24h, i just set this for testing purpose.

/ R

davidusb-geek, 2023-11-01:
This is not a bug. By setting that to True you are just adding more constraint to the optimization problem and it reached infeasible condition.
Your settings are not compatible. Reduce def_total_hours or leave set_def_constant to False

robinostlund, 2023-11-01:
Ahaa oright, will automatically set set_df_constant to false then if 24h :) thanks for the information

closed 2023-11-01 by robinostlund