davidusb-geek/emhass #133: timeslots round to nearest 15 minute

davidsgullick, 2023-11-24:
**Describe the bug**
When using the naive-mpc-optimm it is encouraged to frequently call the function with new data using the endpoint and publish the results. However, when running with a freq of 30 minutes, the results in the Emhass table can sometimes show states starting in the future and not the modelled state for now. 

**To Reproduce**
config set at 30 minutes
Pass data to the naive-mpc-optimm function at a time that would round down to the nearest half hour. For example 9:01 to 9:14 am. This will show in the table timeslots starting at 9:00am onwards. Do the same thing at 9:16 to 9:29 and the timeslots will start at 9:30.

**Expected behavior**
I would expect the first timeslot to always include the current date and time. I think the first slot should be decided by a **floor(CURRENT_TIME)**, not a **round(CURRENT_TIME)**

**Screenshots**
If applicable, add screenshots to help explain your problem.

These screenshots were run one minute apart. The first at 08:14am the second at 08:15am. You can see the first timeslot dissapears in the table (and subsequent publishes to HA)

[image]
[image]

**Home Assistant installation type**
 - Home Assistant Core with Docker

**Your hardware**
- OS: Linux
- Architecture: amd64

**EMHASS installation type**
 - Docker Standalone

**Additional context**

davidusb-geek, 2023-12-19:
Use the parameter `method_ts_round` from the configuration file. You can set the method for timestamp rounding, options are: first, last and nearest.
Don't you think that changing this parameter you may be able to obtain your derired behavior?

davidsgullick, 2023-12-19:
Thanks very much. Yes thats exactly what I needed!

closed 2023-12-19 by davidsgullick