davidusb-geek/emhass #138: Error logger - server not responding

PatrikTrestik, 2023-12-17:
**Describe the bug**
Http endpoint returns ECONRESET for all requests

**To Reproduce**
Not known. Might be the error appeared after HA update.

**Expected behavior**
Normal http response.

**Screenshots**
s6-rc: info: service legacy-services successfully started
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 15, in <module>
    from emhass.command_line import set_input_data_dict
  File "/usr/local/lib/python3.9/dist-packages/emhass/command_line.py", line 20, in <module>
    from emhass.forecast import forecast
  File "/usr/local/lib/python3.9/dist-packages/emhass/forecast.py", line 15, in <module>
    import pvlib
  File "/usr/local/lib/python3.9/dist-packages/pvlib/__init__.py", line 3, in <module>
    from pvlib import (  # noqa: F401
  File "/usr/local/lib/python3.9/dist-packages/pvlib/clearsky.py", line 14, in <module>
    import h5py
  File "/usr/lib/python3/dist-packages/h5py/__init__.py", line 21, in <module>
    from . import _debian_h5py_serial as _h5py
  File "/usr/lib/python3/dist-packages/h5py/_debian_h5py_serial/__init__.py", line 46, in <module>
    from ._conv import register_converters as _register_converters
  File "h5py/_debian_h5py_serial/_conv.pyx", line 1, in init h5py._debian_h5py_serial._conv
  File "h5py/_debian_h5py_serial/h5t.pyx", line 293, in init h5py._debian_h5py_serial.h5t
  File "/usr/local/lib/python3.9/dist-packages/numpy/__init__.py", line 333, in __getattr__
    raise AttributeError("module {!r} has no attribute "
AttributeError: module 'numpy' has no attribute 'typeDict'
s6-supervise emhass: warning: unable to spawn ./finish: Permission denied

**Home Assistant installation type**
 - Home Assistant OS

**Your hardware**
- OS: HA OS
- Architecture:  armv7, aarch64

**EMHASS installation type**
 - Add-on

**Additional context**
Add any other context about the problem here.

nammto, 2023-12-17:
I have the exact same error after EMHASS update.

davidusb-geek, 2023-12-17:
it should be solved, upgrade to add-on v0.5.2 when it is available.
Please close the issue if everything goes back to normal

andreas-bulling, 2023-12-18:
I saw a different error message when trying to install the latest EMHASS version yesterday. Will try 0.5.2 and report back if that version doesn't fix that problem as well.

PatrikTrestik, 2023-12-19:
In new version I see this error when accessing htt endpoint.

2023-12-19 14:20:00,164 - web_server - ERROR - Exception on /action/publish-data [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/indexes/base.py", line 3653, in get_loc
    return self._engine.get_loc(casted_key)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "pandas/_libs/index.pyx", line 147, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/index.pyx", line 176, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/hashtable_class_helper.pxi", line 7080, in pandas._libs.hashtable.PyObjectHashTable.get_item
  File "pandas/_libs/hashtable_class_helper.pxi", line 7088, in pandas._libs.hashtable.PyObjectHashTable.get_item
KeyError: 'optim_status'

davidusb-geek, 2023-12-19:
Ok I also had this same error message myself when installed the new version. You just have to go to the webui and manually update your optimization results. Also while there update the ML forecaster if you are using that. Then manually publish the optimization results. Go back to the logger, the errors should be gone.
This was just because the saved results needed to be updated as we added a new column to the results DataFrame containing the optimization status.

davidusb-geek, 2024-01-04:
Did you manage to apply the solution on my previous post? That should solve it. Closing the issue for now but reopen if needed.

closed 2024-01-04 by davidusb-geek

PatrikTrestik, 2024-01-04:
Hi. Sorry for late answer, I was not home. 
Yes, I can confirm that it helped solve the bug.