davidusb-geek/emhass #139: 0.5.2 unaccepted optimisation behaviour

nammto, 2023-12-18:
Version 0.5.2. Optimisation results are very odd and totally different from 0.4.2. version. 0.4.2. optimisation was working perfect.

**Expected behavior**
As v 0.4.2.

**Screenshots**
v0.4.2 optimisation as expected:
[image]
I don't understand what v0.5.2. is trying to achieve:
[image]

**Home Assistant installation type**
 - Home Assistant OS

**Your hardware**
- OS: HA OS
- HW: rpi4-64

**EMHASS installation type**
 - Add-on

**Additional context**
Restoring back to 0.4.2. and works again as it should.

davidusb-geek, 2024-01-04:
What is the optimization status? These unexpected behavior maybe caused by unfeasible optimization. But it is weird since it work for you on a previous version.

Are you on the latest HAOS version?

Please post the results of: `uname -m` or `dpkg --print-architecture`

nammto, 2024-01-05:
Hi! 
Always running latest HAOS. Updated today also to the latest version and tried EMHASS 0.5.4. Still wierd results.
uname -m gives me aarch64

Found error now in log:
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun emhass (no readiness notification)
s6-rc: info: service legacy-services successfully started
2024-01-05 14:43:35,157 - web_server - INFO - Launching the emhass webserver at: http://0.0.0.0:5000
2024-01-05 14:43:35,157 - web_server - INFO - Home Assistant data fetch will be performed using url: http://supervisor/core/api
2024-01-05 14:43:35,157 - web_server - INFO - The data path is: /share
2024-01-05 14:43:35,162 - web_server - INFO - Using core emhass version: 0.6.2
waitress   INFO  Serving on http://0.0.0.0:5000
2024-01-05 14:44:05,461 - web_server - INFO - EMHASS server online, serving index.html...
2024-01-05 14:44:29,740 - web_server - INFO - Setting up needed data
2024-01-05 14:44:29,816 - web_server - INFO -  >> Publishing data...
2024-01-05 14:44:29,816 - web_server - INFO - Publishing data to HASS instance
2024-01-05 14:44:29,859 - web_server - INFO - Successfully posted to sensor.p_pv_forecast = 12
2024-01-05 14:44:29,883 - web_server - INFO - Successfully posted to sensor.p_load_forecast = 2674.31
2024-01-05 14:44:29,908 - web_server - INFO - Successfully posted to sensor.p_deferrable0 = 0.0
2024-01-05 14:44:29,932 - web_server - INFO - Successfully posted to sensor.p_batt_forecast = 0.0
2024-01-05 14:44:29,957 - web_server - INFO - Successfully posted to sensor.soc_batt_forecast = 83.0
2024-01-05 14:44:29,982 - web_server - INFO - Successfully posted to sensor.p_grid_forecast = 2662.31
2024-01-05 14:44:30,004 - web_server - INFO - Successfully posted to sensor.total_cost_fun_value = -5053.21
2024-01-05 14:44:30,006 - web_server - ERROR - Exception on /action/publish-data [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/indexes/base.py", line 3653, in get_loc
    return self._engine.get_loc(casted_key)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "pandas/_libs/index.pyx", line 147, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/index.pyx", line 176, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/hashtable_class_helper.pxi", line 7080, in pandas._libs.hashtable.PyObjectHashTable.get_item
  File "pandas/_libs/hashtable_class_helper.pxi", line 7088, in pandas._libs.hashtable.PyObjectHashTable.get_item
KeyError: 'optim_status'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 1455, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 869, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 867, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 852, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/web_server.py", line 185, in action_call
    _ = publish_data(input_data_dict, app.logger)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/command_line.py", line 541, in publish_data
    input_data_dict['rh'].post_data(opt_res_latest['optim_status'], idx_closest,
                                   ~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/frame.py", line 3761, in __getitem__
    indexer = self.columns.get_loc(key)
              ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/indexes/base.py", line 3655, in get_loc
    raise KeyError(key) from err
KeyError: 'optim_status'
2024-01-05 14:45:21,871 - web_server - INFO - Setting up needed data
2024-01-05 14:45:21,880 - web_server - INFO - Retrieve hass get data method initiated...
2024-01-05 14:45:31,167 - web_server - INFO - Retrieving weather forecast data using method = list
2024-01-05 14:45:31,173 - web_server - INFO - Retrieving data from hass for load forecast using method = naive
2024-01-05 14:45:31,175 - web_server - INFO - Retrieve hass get data method initiated...
2024-01-05 14:45:50,707 - web_server - INFO -  >> Performing naive MPC optimization...
2024-01-05 14:45:50,707 - web_server - INFO - Performing naive MPC optimization
2024-01-05 14:45:50,733 - web_server - INFO - Perform an iteration of a naive MPC controller
2024-01-05 14:46:32,361 - web_server - INFO - Status: Optimal
2024-01-05 14:46:32,361 - web_server - INFO - Total value of the Cost function = -5056.69
2024-01-05 14:46:55,055 - web_server - INFO - EMHASS server online, serving index.html...
2024-01-05 14:48:27,710 - web_server - INFO - EMHASS server online, serving index.html...

nammto, 2024-01-05:
I did restart EMHASS and now I don't get any error but optimization results are still without any logic.
Log:
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun emhass (no readiness notification)
s6-rc: info: service legacy-services successfully started
2024-01-05 15:03:16,492 - web_server - INFO - Launching the emhass webserver at: http://0.0.0.0:5000
2024-01-05 15:03:16,492 - web_server - INFO - Home Assistant data fetch will be performed using url: http://supervisor/core/api
2024-01-05 15:03:16,493 - web_server - INFO - The data path is: /share
2024-01-05 15:03:16,496 - web_server - INFO - Using core emhass version: 0.6.2
waitress   INFO  Serving on http://0.0.0.0:5000
2024-01-05 15:04:44,795 - web_server - INFO - EMHASS server online, serving index.html...
2024-01-05 15:05:56,530 - web_server - INFO - Setting up needed data
2024-01-05 15:05:56,609 - web_server - INFO -  >> Publishing data...
2024-01-05 15:05:56,609 - web_server - INFO - Publishing data to HASS instance
2024-01-05 15:05:56,680 - web_server - INFO - Successfully posted to sensor.p_pv_forecast = 0
2024-01-05 15:05:56,743 - web_server - INFO - Successfully posted to sensor.p_load_forecast = 2584.62
2024-01-05 15:05:56,782 - web_server - INFO - Successfully posted to sensor.p_deferrable0 = 0.0
2024-01-05 15:05:56,814 - web_server - INFO - Successfully posted to sensor.p_batt_forecast = 7600.0
2024-01-05 15:05:56,846 - web_server - INFO - Successfully posted to sensor.soc_batt_forecast = 55.33
2024-01-05 15:05:56,878 - web_server - INFO - Successfully posted to sensor.p_grid_forecast = -5015.38
2024-01-05 15:05:56,904 - web_server - INFO - Successfully posted to sensor.total_cost_fun_value = -4752.93
2024-01-05 15:05:56,924 - web_server - INFO - Successfully posted to sensor.optim_status = Optimal
2024-01-05 15:05:56,956 - web_server - INFO - Successfully posted to sensor.unit_load_cost = 128.951
2024-01-05 15:05:56,988 - web_server - INFO - Successfully posted to sensor.unit_prod_price = 99.247
2024-01-05 15:06:24,095 - web_server - INFO - Setting up needed data
2024-01-05 15:06:24,103 - web_server - INFO - Retrieve hass get data method initiated...
2024-01-05 15:06:31,924 - web_server - INFO - Retrieving weather forecast data using method = list
2024-01-05 15:06:31,927 - web_server - INFO - Retrieving data from hass for load forecast using method = naive
2024-01-05 15:06:31,928 - web_server - INFO - Retrieve hass get data method initiated...
2024-01-05 15:06:46,947 - web_server - INFO -  >> Performing naive MPC optimization...
2024-01-05 15:06:46,947 - web_server - INFO - Performing naive MPC optimization
2024-01-05 15:06:46,975 - web_server - INFO - Perform an iteration of a naive MPC controller
2024-01-05 15:07:37,993 - web_server - INFO - Status: Optimal
2024-01-05 15:07:37,994 - web_server - INFO - Total value of the Cost function = -4763.26
2024-01-05 15:07:59,690 - web_server - INFO - EMHASS server online, serving index.html... 

Optimization result with old version is okay:
 
[image]

Optimization result with new version is very ugly:
[image]

davidusb-geek, 2024-01-05:
There are two new optional parameters to penalize using the battery. Set them to zero >> 0.0
By default they are set to 1.0, but I will change this default in the next release.

nammto, 2024-01-05:
Hi!

I did set now the values to 0 but no help.
My settings:
hass_url: empty
long_lived_token: empty
costfun: cost
logging_level: INFO
optimization_time_step: 60
historic_days_to_retrieve: 2
method_ts_round: first
set_total_pv_sell: true
lp_solver: COIN_CMD
lp_solver_path: /usr/bin/cbc
set_nocharge_from_grid: false
set_nodischarge_to_grid: false
set_battery_dynamic: false
battery_dynamic_max: 0.9
battery_dynamic_min: -0.9
weight_battery_discharge: 0
weight_battery_charge: 0
load_forecast_method: naive
sensor_power_photovoltaics: sensor.pv_power
sensor_power_load_no_var_loads: sensor.load_power
number_of_deferrable_loads: 1
list_nominal_power_of_deferrable_loads:
  - nominal_power_of_deferrable_loads: 1
  - nominal_power_of_deferrable_loads: 1
list_operating_hours_of_each_deferrable_load:
  - operating_hours_of_each_deferrable_load: 0
  - operating_hours_of_each_deferrable_load: 0
list_peak_hours_periods_start_hours:
  - peak_hours_periods_start_hours: "05:54"
  - peak_hours_periods_start_hours: "05:54"
list_peak_hours_periods_end_hours:
  - peak_hours_periods_end_hours: "05:54"
  - peak_hours_periods_end_hours: "05:54"
list_treat_deferrable_load_as_semi_cont:
  - treat_deferrable_load_as_semi_cont: true
  - treat_deferrable_load_as_semi_cont: true
load_peak_hours_cost: 0
load_offpeak_hours_cost: 0
photovoltaic_production_sell_price: 0
maximum_power_from_grid: 15000
list_pv_module_model:
  - pv_module_model: CSUN_Eurasia_Energy_Systems_Industry_and_Trade_CSUN295_60M
list_pv_inverter_model:
  - pv_inverter_model: Fronius_International_GmbH__Fronius_Primo_5_0_1_208_240__240V_
list_surface_tilt:
  - surface_tilt: 43
list_surface_azimuth:
  - surface_azimuth: 160
list_modules_per_string:
  - modules_per_string: 9
list_strings_per_inverter:
  - strings_per_inverter: 3
set_use_battery: true
battery_discharge_power_max: 8000
battery_charge_power_max: 7000
battery_discharge_efficiency: 0.95
battery_charge_efficiency: 0.92
battery_nominal_energy_capacity: 30000
battery_minimum_state_of_charge: 0.2
battery_maximum_state_of_charge: 1
battery_target_state_of_charge: 0.5

Result different but still thinking I should consume from grid when prices are skyhigh:
[image]

davidusb-geek, 2024-01-05:
What data are you passing to MPC? Why is your `operating_hours_of_each_deferrable_load = 0`?
I don't understand your usecase but from what I can see you have virtually no PV.

nammto, 2024-01-05:
I'm providing nordpool hourly prices, curent soc and final soc, solar forecast data and that's it. I don't have deferrable loads, that's why it's zero. I'm using this pptimisation to charge up battery on lowest hours and usi it for my house supply on the rest of the hours. PV is almost non existent as I have half meter of snow on my panels now :-)

davidusb-geek, 2024-01-08:
Ok, sorry about that snow. I'm currently there myself just right now altough just 1cm, not half meter! ;-)
Ok so I understand your use case better, and I understand your expected behavior from what you got from v0.4.2. The two main thing that changed are the Python version (which can be an issue) and the new constraints to penalize using the battery, setting them to zero should restore your old behavior from v0.4.2. I will need to investigate this further.
Could we please just check your configuration. Could you post again your configuration with the correct formatting? Use the "code" block to post it.

nammto, 2024-01-09:
Hi! 
This is my configuration with v0.4.2:
```
hass_url: empty
long_lived_token: empty
costfun: cost
logging_level: INFO
optimization_time_step: 60
historic_days_to_retrieve: 2
method_ts_round: first
set_total_pv_sell: true
lp_solver: COIN_CMD
lp_solver_path: /usr/bin/cbc
set_nocharge_from_grid: false
set_nodischarge_to_grid: false
set_battery_dynamic: false
battery_dynamic_max: 0.9
battery_dynamic_min: -0.9
load_forecast_method: naive
sensor_power_photovoltaics: sensor.pv_power
sensor_power_load_no_var_loads: sensor.load_power
number_of_deferrable_loads: 1
list_nominal_power_of_deferrable_loads:
  - nominal_power_of_deferrable_loads: 1
  - nominal_power_of_deferrable_loads: 1
list_operating_hours_of_each_deferrable_load:
  - operating_hours_of_each_deferrable_load: 0
  - operating_hours_of_each_deferrable_load: 0
list_peak_hours_periods_start_hours:
  - peak_hours_periods_start_hours: "05:54"
  - peak_hours_periods_start_hours: "05:54"
list_peak_hours_periods_end_hours:
  - peak_hours_periods_end_hours: "05:54"
  - peak_hours_periods_end_hours: "05:54"
list_treat_deferrable_load_as_semi_cont:
  - treat_deferrable_load_as_semi_cont: true
  - treat_deferrable_load_as_semi_cont: true
load_peak_hours_cost: 0
load_offpeak_hours_cost: 0
photovoltaic_production_sell_price: 0
maximum_power_from_grid: 15000
list_pv_module_model:
  - pv_module_model: CSUN_Eurasia_Energy_Systems_Industry_and_Trade_CSUN295_60M
list_pv_inverter_model:
  - pv_inverter_model: Fronius_International_GmbH__Fronius_Primo_5_0_1_208_240__240V_
list_surface_tilt:
  - surface_tilt: 43
list_surface_azimuth:
  - surface_azimuth: 160
list_modules_per_string:
  - modules_per_string: 9
list_strings_per_inverter:
  - strings_per_inverter: 3
set_use_battery: true
battery_discharge_power_max: 8000
battery_charge_power_max: 7000
battery_discharge_efficiency: 0.95
battery_charge_efficiency: 0.92
battery_nominal_energy_capacity: 30000
battery_minimum_state_of_charge: 0.2
battery_maximum_state_of_charge: 1
battery_target_state_of_charge: 0.5
```

And this is with 0.5.4 version:
```
hass_url: empty
long_lived_token: empty
costfun: cost
logging_level: INFO
optimization_time_step: 60
historic_days_to_retrieve: 2
method_ts_round: first
set_total_pv_sell: true
lp_solver: COIN_CMD
lp_solver_path: /usr/bin/cbc
set_nocharge_from_grid: false
set_nodischarge_to_grid: false
set_battery_dynamic: false
battery_dynamic_max: 0.9
battery_dynamic_min: -0.9
weight_battery_discharge: 0
weight_battery_charge: 0
load_forecast_method: naive
sensor_power_photovoltaics: sensor.pv_power
sensor_power_load_no_var_loads: sensor.load_power
number_of_deferrable_loads: 1
list_nominal_power_of_deferrable_loads:
  - nominal_power_of_deferrable_loads: 1
  - nominal_power_of_deferrable_loads: 1
list_operating_hours_of_each_deferrable_load:
  - operating_hours_of_each_deferrable_load: 0
  - operating_hours_of_each_deferrable_load: 0
list_peak_hours_periods_start_hours:
  - peak_hours_periods_start_hours: "05:54"
  - peak_hours_periods_start_hours: "05:54"
list_peak_hours_periods_end_hours:
  - peak_hours_periods_end_hours: "05:54"
  - peak_hours_periods_end_hours: "05:54"
list_treat_deferrable_load_as_semi_cont:
  - treat_deferrable_load_as_semi_cont: true
  - treat_deferrable_load_as_semi_cont: true
load_peak_hours_cost: 0
load_offpeak_hours_cost: 0
photovoltaic_production_sell_price: 0
maximum_power_from_grid: 15000
list_pv_module_model:
  - pv_module_model: CSUN_Eurasia_Energy_Systems_Industry_and_Trade_CSUN295_60M
list_pv_inverter_model:
  - pv_inverter_model: Fronius_International_GmbH__Fronius_Primo_5_0_1_208_240__240V_
list_surface_tilt:
  - surface_tilt: 43
list_surface_azimuth:
  - surface_azimuth: 160
list_modules_per_string:
  - modules_per_string: 9
list_strings_per_inverter:
  - strings_per_inverter: 3
set_use_battery: true
battery_discharge_power_max: 8000
battery_charge_power_max: 7000
battery_discharge_efficiency: 0.95
battery_charge_efficiency: 0.92
battery_nominal_energy_capacity: 30000
battery_minimum_state_of_charge: 0.2
battery_maximum_state_of_charge: 1
battery_target_state_of_charge: 0.5
```

davidusb-geek, 2024-01-09:
Test with this:

```
hass_url: empty
long_lived_token: empty
costfun: cost
logging_level: INFO
optimization_time_step: 60
historic_days_to_retrieve: 2
method_ts_round: first
set_total_pv_sell: true
lp_solver: COIN_CMD
lp_solver_path: /usr/bin/cbc
set_nocharge_from_grid: false
set_nodischarge_to_grid: false
set_battery_dynamic: false
battery_dynamic_max: 0.9
battery_dynamic_min: -0.9
weight_battery_discharge: 0.0
weight_battery_charge: 0.0
load_forecast_method: naive
sensor_power_photovoltaics: sensor.pv_power
sensor_power_load_no_var_loads: sensor.load_power
number_of_deferrable_loads: 1
list_nominal_power_of_deferrable_loads:
  - nominal_power_of_deferrable_loads: 1.0
list_operating_hours_of_each_deferrable_load:
  - operating_hours_of_each_deferrable_load: 1
list_peak_hours_periods_start_hours:
  - peak_hours_periods_start_hours: "05:54"
list_peak_hours_periods_end_hours:
  - peak_hours_periods_end_hours: "05:54"
list_treat_deferrable_load_as_semi_cont:
  - treat_deferrable_load_as_semi_cont: false
load_peak_hours_cost: 0
load_offpeak_hours_cost: 0
photovoltaic_production_sell_price: 0
maximum_power_from_grid: 15000
list_pv_module_model:
  - pv_module_model: CSUN_Eurasia_Energy_Systems_Industry_and_Trade_CSUN295_60M
list_pv_inverter_model:
  - pv_inverter_model: Fronius_International_GmbH__Fronius_Primo_5_0_1_208_240__240V_
list_surface_tilt:
  - surface_tilt: 43
list_surface_azimuth:
  - surface_azimuth: 160
list_modules_per_string:
  - modules_per_string: 9
list_strings_per_inverter:
  - strings_per_inverter: 3
set_use_battery: true
battery_discharge_power_max: 8000
battery_charge_power_max: 7000
battery_discharge_efficiency: 0.95
battery_charge_efficiency: 0.92
battery_nominal_energy_capacity: 30000
battery_minimum_state_of_charge: 0.2
battery_maximum_state_of_charge: 1
battery_target_state_of_charge: 0.5
```

When you launch this do you have any message related to the solver in the logs?

nammto, 2024-01-09:
Hi!
No errors in the log:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun emhass (no readiness notification)
s6-rc: info: service legacy-services successfully started
2024-01-09 12:39:23,035 - web_server - INFO - Launching the emhass webserver at: http://0.0.0.0:5000
2024-01-09 12:39:23,035 - web_server - INFO - Home Assistant data fetch will be performed using url: http://supervisor/core/api
2024-01-09 12:39:23,035 - web_server - INFO - The data path is: /share
2024-01-09 12:39:23,039 - web_server - INFO - Using core emhass version: 0.6.2
waitress   INFO  Serving on http://0.0.0.0:5000
2024-01-09 12:40:34,878 - web_server - INFO - Setting up needed data
2024-01-09 12:40:34,955 - web_server - INFO - Retrieve hass get data method initiated...
2024-01-09 12:40:42,570 - web_server - INFO - Retrieving weather forecast data using method = list
2024-01-09 12:40:42,573 - web_server - INFO - Retrieving data from hass for load forecast using method = naive
2024-01-09 12:40:42,575 - web_server - INFO - Retrieve hass get data method initiated...
2024-01-09 12:41:05,193 - web_server - INFO -  >> Performing naive MPC optimization...
2024-01-09 12:41:05,194 - web_server - INFO - Performing naive MPC optimization
2024-01-09 12:41:05,213 - web_server - INFO - Perform an iteration of a naive MPC controller
2024-01-09 12:41:05,331 - web_server - INFO - Status: Optimal
2024-01-09 12:41:05,331 - web_server - INFO - Total value of the Cost function = -461.53
2024-01-09 12:41:06,514 - web_server - INFO - Setting up needed data
2024-01-09 12:41:06,519 - web_server - INFO -  >> Publishing data...
2024-01-09 12:41:06,519 - web_server - INFO - Publishing data to HASS instance
2024-01-09 12:41:06,558 - web_server - INFO - Successfully posted to sensor.p_pv_forecast = 16
2024-01-09 12:41:06,583 - web_server - INFO - Successfully posted to sensor.p_load_forecast = 1188.04
2024-01-09 12:41:06,606 - web_server - INFO - Successfully posted to sensor.p_deferrable0 = 0.0
2024-01-09 12:41:06,635 - web_server - INFO - Successfully posted to sensor.p_batt_forecast = 7600.0
2024-01-09 12:41:06,662 - web_server - INFO - Successfully posted to sensor.soc_batt_forecast = 47.33
2024-01-09 12:41:06,692 - web_server - INFO - Successfully posted to sensor.p_grid_forecast = -6427.96
2024-01-09 12:41:06,714 - web_server - INFO - Successfully posted to sensor.total_cost_fun_value = -461.53
2024-01-09 12:41:06,739 - web_server - INFO - Successfully posted to sensor.optim_status = Optimal
2024-01-09 12:41:06,765 - web_server - INFO - Successfully posted to sensor.unit_load_cost = 21.603
2024-01-09 12:41:06,788 - web_server - INFO - Successfully posted to sensor.unit_prod_price = 9.79
2024-01-09 12:41:49,585 - web_server - INFO - EMHASS server online, serving index.html...
```
But optimisation still wrong :
[image]

nammto, 2024-01-09:
I add also the same log and optimisations result with ver 0.4.2:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun emhass (no readiness notification)
s6-rc: info: service legacy-services successfully started
2024-01-09 12:47:42,005 - web_server - INFO - Launching the emhass webserver at: http://0.0.0.0:5000
2024-01-09 12:47:42,006 - web_server - INFO - Home Assistant data fetch will be performed using url: http://supervisor/core/api
2024-01-09 12:47:42,006 - web_server - INFO - The data path is: /share
2024-01-09 12:47:42,011 - web_server - INFO - Using core emhass version: 0.5.1
waitress   INFO  Serving on http://0.0.0.0:5000
2024-01-09 12:48:05,874 - web_server - INFO - Setting up needed data
2024-01-09 12:48:05,953 - web_server - INFO - Retrieve hass get data method initiated...
2024-01-09 12:48:13,524 - web_server - INFO - Retrieving weather forecast data using method = list
2024-01-09 12:48:13,528 - web_server - INFO - Retrieving data from hass for load forecast using method = naive
2024-01-09 12:48:13,529 - web_server - INFO - Retrieve hass get data method initiated...
2024-01-09 12:48:31,064 - web_server - INFO -  >> Performing naive MPC optimization...
2024-01-09 12:48:31,065 - web_server - INFO - Performing naive MPC optimization
2024-01-09 12:48:31,088 - web_server - INFO - Perform an iteration of a naive MPC controller
2024-01-09 12:48:31,210 - web_server - INFO - Status: Optimal
2024-01-09 12:48:31,210 - web_server - INFO - Total value of the Cost function = -140.95
2024-01-09 12:48:32,422 - web_server - INFO - Setting up needed data
2024-01-09 12:48:32,428 - web_server - INFO -  >> Publishing data...
2024-01-09 12:48:32,428 - web_server - INFO - Publishing data to HASS instance
2024-01-09 12:48:32,470 - web_server - INFO - Successfully posted to sensor.p_pv_forecast = 16
2024-01-09 12:48:32,495 - web_server - INFO - Successfully posted to sensor.p_load_forecast = 1188.04
2024-01-09 12:48:32,524 - web_server - INFO - Successfully posted to sensor.p_deferrable0 = 0.0
2024-01-09 12:48:32,553 - web_server - INFO - Successfully posted to sensor.p_batt_forecast = 1172.04
2024-01-09 12:48:32,584 - web_server - INFO - Successfully posted to sensor.soc_batt_forecast = 70.89
2024-01-09 12:48:32,615 - web_server - INFO - Successfully posted to sensor.p_grid_forecast = 0.0
2024-01-09 12:48:32,641 - web_server - INFO - Successfully posted to sensor.total_cost_fun_value = -461.53
2024-01-09 12:48:32,667 - web_server - INFO - Successfully posted to sensor.unit_load_cost = 21.603
2024-01-09 12:48:32,692 - web_server - INFO - Successfully posted to sensor.unit_prod_price = 9.79
```
[image]

nammto, 2024-01-10:
Hi again!
One more note. I have very similar HA setup at my fathers house also and now tested the new version there and it's misbehaving exactly as in my home setup.

davidusb-geek, 2024-01-10:
Hi. Ok I will take a more in-depth look at this this week-end and try to reproduce your behavior. I will keep you updated.

davidusb-geek, 2024-01-22:
I began trying to replicate your issue using your config this weekend. Still not there, got some issues computing the final cost. I just need to spend a little more time with this. But in the process I noticed that all your costs are 0, so how could we optimize something if all the fixed costs are 0?

nammto, 2024-01-23:
Hi! Thanks for still dealing with that. I'm forwarding all cost data with MPC. I have all the costs calculated beforehand.
Example of MPC post:
`post_mpc_optim: "curl -i -H \"Content-Type: application/json\" -X POST -d '{
        \"load_cost_forecast\":[17.836, 19.146, 18.753, 17.838, 17.277, 16.282, 16.736, 16.047, 17.004, 19.982, 17.17, 16.968, 16.556, 16.21, 12.333, 10.937], 
        \"prod_price_forecast\":[6.651, 7.743, 7.415, 6.653, 6.185, 5.356, 5.734, 5.16, 5.958, 8.439, 6.096, 5.928, 5.584, 5.296, 4.495, 3.332],
        \"prediction_horizon\":16,
        \"pv_power_forecast\": [0, 8, 27, 42, 47, 41, 25, 7, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 19, 52, 73, 74, 68, 44, 12, 0, 0, 0, 0],
        \"alpha\": 1,
        \"beta\": 0,
        \"soc_init\":0.98,
        \"soc_final\":0.3,
        \"def_total_hours\":[0]       
      }' http://localhost:5000/action/naive-mpc-optim"`

davidusb-geek, 2024-01-27:
Ok I think that I found the issue!
Set parameter `set_total_pv_sell` to `False`.
This behavior changed in the new version as a consistency error was found in the old code.
Test it to see if it is the results that you expect.

nammto, 2024-01-28:
Hi! Thank You very much, it worked!

closed 2024-01-28 by nammto