davidusb-geek/emhass #141: Type of self-consumption optimization is always bigm, never maxmin

michaelpiron, 2023-12-27:
Hi @davidusb-geek ,

In the code, I noticed that in case of a self-consumption cost function, the optimizer always applies the so-called "bigm" method, due to this hard-coded parameter:

https://github.com/davidusb-geek/emhass/blob/a4255f54f2c5ae2e9e34b0fd5ee6d7c1ed2efef6/src/emhass/optimization.py#L138

However, the documentation explains the "maxmin" method, which is never invoked due to the parameter above.
Is there an issue with the maxmin method (e.g. never leads to a solution)?

Happy to help further, based on your reply.

davidusb-geek, 2023-12-27:
Yes it is hard coded to big M. It was long time ago but I think that the other method has convergence problems, probably the equations are not well posed. The big M was a second possible method but the maxmin is the elegant correct way to handle this. 
Of course that you can help, all contributions are more than welcomed.
If you are interested we have another problem in the constraints section.

michaelpiron, 2023-12-27:
Ok, I will adapt the documentation to reflect the reality that self-consumption cost function uses the big M method.

regarding the other issue on constraints, is there already a bug report of this? If not, can you provide more input/evidence of the issue?

thx

closed 2024-01-04 by davidusb-geek

michaelpiron, 2024-01-04:
Updated the optimizer documentation here: #143 
It now describes that self-consumption has two methods (bigM and maxmin), and that currently only method bigM is used.