davidusb-geek/emhass #144: load_cast_forecast nordpool fails

mschneider82, 2024-01-04:
Hi,

i tried to load the data from nordpool forecast from documentation but i get the failure:

```
2024-01-04 13:09:10,166 - web_server - ERROR - ERROR: The passed data is either not a list or the length is not correct, length should be 48
2024-01-04 13:09:10,167 - web_server - ERROR - Passed type is <class 'list'> and length is 24
```

my post command:
```
  trigger_nordpool_forecast: "curl -i -H \"Content-Type: application/json\" -X POST -d '{\"load_cost_forecast\":{{((state_attr('sensor.nordpool_kwh_de_lu_eur_3_095_0', 'raw_today') | map(attribute='value') | list  + state_attr('sensor.nordpool_kwh_de_lu_eur_3_095_0', 'raw_tomorrow') | map(attribute='value') | list))[now().hour:][:24] }}}' http://emhass:5000/action/dayahead-optim"
  ```
  
  appling the template in the editor gets this values:
  
 Result type: string
[0.29, 0.297, 0.306, 0.339, 0.352, 0.329, 0.303, 0.299, 0.293, 0.286, 0.283, 0.276, 0.272, 0.267, 0.268, 0.271, 0.277, 0.284, 0.287, 0.293, 0.295, 0.297, 0.29, 0.289]}

I have skipped the "prod_price_forecast" from the docs in the curl because we have static pv export prices in germany.
Why it expects 48 values? We have just prices per hour.

ronaldt80, 2024-01-04:
Is your time step set to 60? If its still at 30 (default value) the it expects 48 values for a 24 hour period

mschneider82, 2024-01-04:
Do you mean this one:
```
retrieve_hass_conf:
  - freq: 30 # The time step to resample retrieved data from hass in minutes
```

mschneider82, 2024-01-04:
I have changed it to 60 and it works, thanks, maybe we should add this to the nordpool configuration docs
thanks!

closed 2024-01-04 by mschneider82