davidusb-geek/emhass #145: Bug: Incorrect list of supported modules / inverters in docs

ha-enthus1ast, 2024-01-08:
Hi,

According to the [the documentation](https://emhass.readthedocs.io/en/latest/config.html#system-configuration-parameters) the list of modules / inverters supported are found [here](https://github.com/davidusb-geek/emhass-add-on/files/9234460/sam-library-cec-modules-2019-03-05.csv) and [here](https://github.com/davidusb-geek/emhass-add-on/files/9532724/sam-library-cec-inverters-2019-03-05.csv).

However, as far as I can tell inverter/modules are retrieved from the oline SAM database. See snippet below

https://github.com/davidusb-geek/emhass/blob/56913605ad36a4b674c294c48a90ea1d2a6fdb7b/src/emhass/forecast.py#L413

Which indicates, according to [pvlib docs](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.pvsystem.retrieve_sam.html) that the online database is used.

The actual list of supported modules are located [here](https://github.com/NREL/SAM/tree/develop/deploy/libraries)

Thanks in advance

ha-enthus1ast, 2024-01-08:
Oh, you are correct. That was very suprising and I think the pvlib documentation is insufficient in this case.

However, looking at the pvlib source you linked, it should be possible to supply your own cvc file, right? ie by only passing a path to a cvc file, rather that passing a name.

`cec_inverters = pvlib.pvsystem.retrieve_sam('<path_to_custom_cvc') `

I am not stuck in any way I was just confused on which file was actually used.

Thanks for getting back to me!

davidusb-geek, 2024-01-08:
Yes if we can put together un updated version of the CSV file then maybe we can pass it directly yes. it can be an option. Now where can we find this updated version. We need to keep the same formatting as the current CSV file.

ha-enthus1ast, 2024-01-08:
As I mentioned in the issue, pvlib refers to files [here](https://github.com/NREL/SAM/tree/develop/deploy/libraries) where it seems the latest update of the list is for fall 2023. It seems to me that the format of the inverter file has somewhat changed, but the one for the modules seems to be at least similar. Whether any these potential format changes are supported by pvlib or not i do not know..

ha-enthus1ast, 2024-01-08:
This approach would however require someone to manually update the csv file on new lists. It could of course be retreived from the source by supplying a URL instead, but that is probably a bad idea since it would rely on an internet connection and on the fact that pvlib supports all future formats of the file.

davidusb-geek, 2024-01-09:
I didn't saw that option to pass a CSV path and from what I can see that updated CSV has the same format except for the last column. I think that we could use that. I will try it and if it works I will include it in the next release.

ha-enthus1ast, 2024-01-09:
Cool thanks!

ha-enthus1ast, 2024-01-09:
Closing issue as invalid, see this [comment](https://github.com/davidusb-geek/emhass/issues/145#issuecomment-1881118037)

davidusb-geek, 2024-02-04:
I reopened this issue because we could possible update the CSV files stored in EMHASS to thos more updated files from SAM. But this need to be tested.

davidusb-geek, 2024-03-31:
I made the needed changes for this. It will be available in the following releases.
Here is a post in the community forum: https://community.home-assistant.io/t/emhass-an-energy-management-for-home-assistant/338126/2031?u=davidusb
And the link to the new webapp to help search the modules and inverter names for your configuration: https://emhass-pvlib-database.streamlit.app/

davidusb-geek, 2024-04-01:
Released on latest version.
Closing

closed 2024-04-01 by davidusb-geek