davidusb-geek/emhass #152: Feature request: round price values to 3rd decimal place instead of 2nd

g1za, 2024-01-23:
Hello,
I think it could be useful to see all the nuances and have wider flexibility if we could round the prices to no less than the **third decimal place**.
For example I have a 3 level tariff and often the two most expensive can't be distinguished hence the optimization is not fully perfect in such situation.
Thank you!

For example this is my load cost forecast for the next hours:
`[
  0.134031,
  0.134031,
  0.134031,
  0.134031,
  0.134031,
  0.134031,
  0.134031,
  0.134031,
  0.128332,
  0.128332,
  0.128332,
  0.128332,
  0.128332,
  0.128332,
  0.128332,
  0.128332,
  0.111448,
  0.111448,
  0.111448,
  0.111448,
  0.111448,
  0.111448,
  0.111448,
  0.111448,
  0.111448,
  0.111448,
  0.111448,
  0.111448,
  0.111448,
  0.111448,
  0.111448,
  0.111448,
  0.128332,
  0.128332,
  0.134031,
  0.134031,
  0.134031,
  0.134031,
  0.134031,
  0.134031,
  0.134031,
  0.134031,
  0.134031,
  0.134031,
  0.134031,
  0.134031,
  0.134031,
  0.134031
]`
and this is what you it seems you get following the optimization and what drives it
`0.13
0.13
0.13
0.13
0.13
0.13
0.13
0.13
0.13
0.13
0.13
0.13
0.13
0.13
0.13
0.13
0.11
0.11
0.11
0.11
0.11
0.11
0.11
0.11
0.11
0.11
0.11
0.11
0.11
0.11
0.11
0.11
0.13
0.13
0.13
0.13
0.13
0.13
0.13
0.13
0.13
0.13
0.13
0.13
0.13
0.13
0.13
0.13
`

davidusb-geek, 2024-01-23:
No, there is no rounding involved when supplying data to EMHASS. 
Rounding is done only to post the results table on the webui.

g1za, 2024-01-23:
Ah ok. So I assume the same is applicable to plots as well.
Closing the ticket then.
Thank you!

closed 2024-01-23 by g1za

g1za, 2024-02-07:
I see you changed this behavior both in charts and the table and now numbers are rounded to the 3rd decimal.
Appreciated.