davidusb-geek/emhass #155: Use time_zone (or new param) to calculate local currency symbol

GeoDerp, 2024-01-25:
https://github.com/davidusb-geek/emhass/blob/3dfcf1b4032fc0677b489049a98e93581218b44a/src/emhass/utils.py#L140-L141

I could see this having multiple solutions. 
- new pram that just has the currency sign 
  - `currency="$"` 
 - parse the time_zone param using an python library 
 - new param that overrides locale country code, otherwise let local guess 
 ```python
 .
 .
.
import locale

if CountryCode != "":
    locale.setlocale(locale.LC_ALL, CountryCode)
else:
    locale.setlocale(locale.LC_ALL, '')

currency_symbol = locale.localeconv()["int_curr_symbol"] 
.
.
.
 'custom_unit_load_cost_id': {"entity_id": "sensor.unit_load_cost", "unit_of_measurement": "currency_symbol", "friendly_name": "Unit Load Cost"}, 
 'custom_unit_prod_price_id': {"entity_id": "sensor.unit_prod_price", "unit_of_measurement": "currency_symbol", "friendly_name": "Unit Prod Price"},
 .
 .
 .
 ```

davidusb-geek, 2024-01-25:
What is the problem with the existent?
There is a method to pass your own custom names, currency, etc

GeoDerp, 2024-01-25:
ah my bad.

davidusb-geek, 2024-01-25:
Yes no problem. Those lines that you posted are just there as default values but the user can change to whatever they want.

GeoDerp, 2024-01-27:
@davidusb-geek  is the "currency"  homeassistant variable being passed in. Could that be used as default?

davidusb-geek, 2024-01-27:
No, currency is not being fetched from HA. It could be interesting to fetch it directly yes.

GeoDerp, 2024-01-27:
A temporary stop gap could be another param *(for addon users)*:

web_server.py 
```python
...
        params['retrieve_hass_conf']['alt'] = options['Altitude']
        params['retrieve_hass_conf']['unit_of_measurement']  =  options['Currency']
        params['optim_conf']['set_use_battery'] = options['set_use_battery']
...        
```

utils.py
```python
...
if 'custom_unit_load_cost_id' in runtimeparams.keys():
            params['passed_data']['custom_unit_load_cost_id'] = runtimeparams['custom_unit_load_cost_id']
        elif 'unit_of_measurement' in params['retrieve_hass_conf']:
            params['passed_data']['custom_unit_load_cost_id']['unit_of_measurement'] = str(params['retrieve_hass_conf']['unit_of_measurement']+"/kWh") 
        if 'custom_unit_prod_price_id' in runtimeparams.keys():
            params['passed_data']['custom_unit_prod_price_id'] = runtimeparams['custom_unit_prod_price_id']
        elif 'unit_of_measurement' in params['retrieve_hass_conf']:
            params['passed_data']['custom_unit_prod_price_id']['unit_of_measurement'] = str(params['retrieve_hass_conf']['unit_of_measurement']+"/kWh")
...
```

davidusb-geek, 2024-03-11:
To implement this, there should not be any stop gap for add-on users or any other installation method.
Actually if we do:
```
url = self.hass_url+"api/config"
response = get(url, headers=headers)
response.json()['currency']
```
In my case I obtain `'EUR'`. So we actually have access to all the configuration parameters directly retrieved from HA.

davidusb-geek, 2024-10-29:
So @GeoDerp do you think that it is possible to fetch the currency directly from the HA configuration?

purcell-lab, 2024-10-29:
My current work around for this is:

```yaml
  publish_data:
    url: http://localhost:5000/action/publish-data
    method: POST
    content_type: "application/json"
    payload: '{"custom_unit_load_cost_id": {"entity_id": "sensor.unit_load_cost", "unit_of_measurement": "$/kWh", "friendly_name": "Load Cost"}, "custom_unit_prod_price_id": {"entity_id": "sensor.unit_prod_price", "unit_of_measurement": "$/kWh", "friendly_name": "Prod Price"}}'
```

I would welcome it automatically pulling the locale currency.

davidusb-geek, 2025-01-06:
Support for this was added on PR #399 
Closing as completed

closed 2025-01-06 by davidusb-geek