davidusb-geek/emhass #156: Feature Request: allow variable power per timestep

sti0, 2024-01-27:
I want to request a new feature which could make the usage of EMHASS more efficient. 

Let's assume we want to schedule a dishwasher which consumes 2000W at peak. The programm duration is 3h. The device does not consume 2000W the whole 3 hours. As we know the most power is consumed in the first or maybe second timestep (30m timeframe) to heat the water. Maybe there is another higher consumption at the end when it dries.

It would be nice if we could add a list of power according to the defined hours. In case of the dishwasher the list could look like this: [2000,2000,250,250,1000,50]

davidusb-geek, 2024-01-27:
Yes this could be nice, good idea.
Open to contributions to implement this.

purcell-lab, 2024-01-30:
> It would be nice if we could add a list of power according to the defined hours. In case of the dishwasher the list could look like this: [2000,2000,250,250,1000,50]

This would be good.

This is what my dishwasher profile looks like:
[image]

davidusb-geek, 2024-01-30:
Yes this is interesting.
However my original reasoning for implementing as it is today was this. We provide just one nominal power and the number of functioning hours for our deferrable load, here the dishwasher. So with this information we are providing the number of total kWh (power*hours) that this deferrable load needs for its cycle. 
For someone starting with no prior full knowlege of the deferrable load profile that reasoning is easier to implement.
However if you have a meter and fully know you load cycle power profile, then yes it makes sense to provide the algorithm with this more detailed information.
The downside as always is that this will add more constraint to the optimization problem. With the risk that our open source solver won't be able to handle such a big problem if you have for example several deferrable loads (more than 2 I will say). While we can contour dimensionality problem with MPC for example, we are still prone to the curse of dimensionality (https://en.wikipedia.org/wiki/Curse_of_dimensionality)