davidusb-geek/emhass #186: Reload page/diagram on button press

GeoDerp, 2024-02-06:
I can see this being done either via python as a response *(to completing the task)* or via JavaScript to refresh after X time after button press. 
I believe it would be nice to have the latest data be presented on the webpage after pressing the respective button in the EMHASS webpage. The implementation of this statically would be the easiest. Telling the whole page to refresh. Ideally I would believe the best method would be refreshing the diagram div's dynamically on new data.

davidusb-geek, 2024-02-06:
That would be nice.
While you are at this take also a look at this if you can: https://github.com/users/davidusb-geek/projects/1?pane=issue&itemId=52141646

GeoDerp, 2024-02-08:
Had a 2 second look at this but I believe if we can redraw the {{injection_dict[plot]}} on every change or X seconds we might be able to the diagram to dynamically refresh. 🤷‍♂️

davidusb-geek, 2024-02-08:
Ok, I think should be the best option, but how to redraw on every change on web_server.py ?

GeoDerp, 2024-02-08:
That I don't know. We might be able to passively scan for changes to the file. However I haven't observed the code to see if redrawing the loop is even possible.

GeoDerp, 2024-02-09:
I'm wondering if, just before the response of a /action button, we call a function to rerun this?

```python
        injection_dict={}
    basename = request.headers.get("X-Ingress-Path", "")
    return make_response(template.render(injection_dict=injection_dict, basename=basename))
```
If this response is even responsible for drawing the diagrams. Only has a look at the code for a few mins so far

GeoDerp, 2024-02-11:
maybe something like this? 
https://stackoverflow.com/a/16881055

after button response we run a http get to get the latest html of the graphs and replace (using JS)

davidusb-geek, 2024-02-25:
Is this solved with the new version?

GeoDerp, 2024-02-25:
Sure is 👍

closed 2024-02-25 by GeoDerp