davidusb-geek/emhass #187: ZeroDivisionError  on dayahead-optim

mschneider82, 2024-02-06:
**Describe the bug**

i have an automatic trigger for trigger_nord_pool_forecast that always fails, when i run shell_command.trigger_nordpool_forecast manual it works.

```
  - alias: EMHASS day-ahead nordpool
    trigger:
      platform: time
      at: '14:30:00'
    action:
    - service: shell_command.trigger_nordpool_forecast
```

This is the exception i get one minute after that trigger:

```
ZeroDivisionError: integer division or modulo by zero
            ~~~~~~~~~^~~~~~~~~~~
    e = b + (iend - b) // stride * stride + stride // 2 + 1
  File "/usr/local/lib/python3.11/site-packages/pandas/core/arrays/_ranges.py", line 68, in generate_regular_range
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    i8values = generate_regular_range(start, end, periods, freq, unit=unit)
  File "/usr/local/lib/python3.11/site-packages/pandas/core/arrays/datetimes.py", line 446, in _generate_range
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    dtarr = DatetimeArray._generate_range(
  File "/usr/local/lib/python3.11/site-packages/pandas/core/indexes/datetimes.py", line 945, in date_range
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    dti = date_range(obj.index.min(), obj.index.max(), freq=freq)
  File "/usr/local/lib/python3.11/site-packages/pandas/core/resample.py", line 2232, in asfreq
           ^^^^^^^
    return asfreq(
  File "/usr/local/lib/python3.11/site-packages/pandas/core/generic.py", line 8347, in asfreq
           ^^^^^^^^^^^^^^^
    return super().asfreq(
  File "/usr/local/lib/python3.11/site-packages/pandas/core/frame.py", line 10971, in asfreq
         ^^^^^^^^^^^^^^^^^^^
    df = df.asfreq(sampling)
  File "/usr/local/lib/python3.11/site-packages/emhass-0.7.2-py3.11.egg/emhass/utils.py", line 720, in set_df_index_freq
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    df_input_data_dayahead = utils.set_df_index_freq(df_input_data_dayahead)
  File "/usr/local/lib/python3.11/site-packages/emhass-0.7.2-py3.11.egg/emhass/command_line.py", line 95, in set_input_data_dict
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    input_data_dict = set_input_data_dict(config_path, str(data_path), costfun,
  File "/app/src/emhass/web_server.py", line 50, in action_call
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
  File "/usr/local/lib/python3.11/site-packages/flask/app.py", line 855, in dispatch_request
         ^^^^^^^^^^^^^^^^^^^^^^^
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.11/site-packages/flask/app.py", line 870, in full_dispatch_request
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.11/site-packages/flask/app.py", line 872, in full_dispatch_request
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.11/site-packages/flask/app.py", line 1463, in wsgi_app
Traceback (most recent call last):
2024-02-05 13:31:05,392 - web_server - ERROR - Exception on /action/dayahead-optim [POST]

```

davidusb-geek, 2024-02-06:
And what is your `shell_command.trigger_nordpool_forecast`?
What does the EMHASS logs says just before this error message?

mschneider82, 2024-02-07:
i think this was triggered by the solcast not the nordpool forecast. 
I had two triggers, i will combine them into one dayahead-optim call. But maybe the division by zero should be fixed.

closed 2024-02-07 by mschneider82