davidusb-geek/emhass #188: I am wondering if this would be approved by @davidusb-geek? 
This would take alot of time to create but the result would be far greater then what we can make with the limitations of the HA configuration panel.

GeoDerp, 2024-02-07:
Another idea I had *(which would take a while to make)* is a html form that would override the need for the add-on configuration tab *(the configuration tab would only have url and key)*. This could be formatted in a way that makes parameters readable, sectioned and also dynamically hidden when certain workloads/parameters are defined.

On submission emhass could set each parameters into a option.json but could also create the necessary inputs/sensors/automations etc *(restrictions based on what we can do with the HA API)*

_Originally posted by @GeoDerp in https://github.com/davidusb-geek/emhass-add-on/pull/73#issuecomment-1926121430_

GeoDerp, 2024-02-07:
Just a note: I may take a hiatus from working on this project after the current pr's are finished.

davidusb-geek, 2024-02-07:
> _Originally posted by @GeoDerp in [davidusb-geek/emhass-add-on#73 (comment)](https://github.com/davidusb-geek/emhass-add-on/pull/73#issuecomment-1926121430)_
> 
> I wonder if this is kinda related to: https://github.com/users/davidusb-geek/projects/1?pane=issue&itemId=52141646

For me it is not related. What I put on github projet is just to add a single text input box where anyone can put their passed data dictionary. The one that we pass using the `curl` command for runtime data. So that when we press the button for optimization in the webui this dictionary is passed along with the POST request. This shouldn't be hard to do but I lack the skills on this.

What you are proposing is about the configuration to be set up on the webui. For now is hard for me to see the need to supplant the configuration tab from the add-on. I guess that this can be useful in docker standalone mode?

davidusb-geek, 2024-02-07:
> Just a note: I may take a hiatus from working on this project after the current pr's are finished.

Thanks again for all the work you have done

GeoDerp, 2024-02-07:
@davidusb-geek , feel free to assign  this to me:
https://github.com/users/davidusb-geek/projects/1?pane=issue&itemId=52141646

Arround half way through implementing this. Will test the back end when i'm free next and update you on progress.

GeoDerp, 2024-02-07:
@davidusb-geek , If you got some good examples/parameters I could test with let me know.

davidusb-geek, 2024-02-07:
I somehow erased the original project tab for that task.
Here is the new one: https://github.com/users/davidusb-geek/projects/1/views/1?pane=issue&itemId=52545404

So the idea is to create a text box where we can pass the data `dict` with runtime params.
So that when we push the dayahead optimization button we pass the data in the text box.
For example the data in the box could be this dictionary:
```
{"pv_power_forecast":[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 70, 141.22, 246.18, 513.5, 753.27, 1049.89, 1797.93, 1697.3, 3078.93, 1164.33, 1046.68, 1559.1, 2091.26, 1556.76, 1166.73, 1516.63, 1391.13, 1720.13, 820.75, 804.41, 251.63, 79.25, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],"load_power_forecast":[459, 1197, 1532, 1701, 1860, 167, 1878, 603, 823, 129, 1752, 1717, 1568, 1177, 732, 755, 1793, 865, 1143, 592, 1078, 197, 102, 317, 1058, 1310, 520, 1671, 247, 226, 1297, 214, 1707, 280, 787, 2011, 1714, 960, 1359, 1200, 389, 77, 861, 940, 678, 277, 473, 1667]}
```
Then when we push the button this dictionary is passed to the following POST call inside the `-d {}` argument:
```
curl -i -H 'Content-Type:application/json' -X POST -d '{}' http://localhost:5000/action/dayahead-optim
```

GeoDerp, 2024-02-08:
> I somehow erased the original project tab for that task. Here is the new one: https://github.com/users/davidusb-geek/projects/1/views/1?pane=issue&itemId=52545404
>
> So the idea is to create a text [...]

PR #190 👍. hope you like it

closed 2024-03-03 by davidusb-geek