davidusb-geek/emhass #191: Return action error logs to webserver

GeoDerp, 2024-02-11:
In its current status the response message to the action is a status code Ex: https://github.com/davidusb-geek/emhass/blob/78309ad7b2a8f6f280deadec4bca378279476d00/src/emhass/web_server.py#L55
This however doesn't indicate if there was an error during action (and if there was, what was it). This only gives us the information that the action has completed 

 PR #190 implements alert box element that currently only indicates if there has been a JSON input error:
https://github.com/davidusb-geek/emhass/blob/78309ad7b2a8f6f280deadec4bca378279476d00/src/emhass/templates/index.html#L200-L203

I believe we could modify the `return make_response(msg, 201)` to send back a 400 code with an error log if something went wrong. we could then tell the JS to present the html alert box with the response error data. *(helping the user understand why it crashed without viewing the logs)*

js example being:
```js
    async function formAction(action) {
        var data = inputToJson()
        if (data !== 0) { //don't run if there is an error in the data Json
            showChangeStatus("loading") // show loading div for status
            response = await fetch(`{{ basename }}/action/${action}`, {
                method: "POST",
                headers: {
                    "Content-Type": "application/json",
                },
                body: JSON.stringify(data), //post can only send data via strings
            }).then(saveStorage()) //save to storage if successful 
            showChangeStatus(response.status) //replace loading, show tick or cross
        }
        else {
            showChangeStatus("remove") //replace loading, show tick or cross with none 
        }
    }

    //function in control of status icons of post above
    function showChangeStatus(status) {
        var loading = document.getElementById("loader") //element showing status 
        if (status === "remove") { //show loading logo
            loading.innerHTML = "";
            loading.classList.remove("loading"); //append class with loading animation styling
        }
        else if (status === "loading") { //show loading logo
            loading.innerHTML = "";
            loading.classList.add("loading"); //append class with loading animation styling
        }
        else if (status === 201) { //then show a tick 
            loading.classList.remove("loading")
            loading.innerHTML = `<p class=tick>&#x2713;</p>`

        }
        else { //then show a cross 
            loading.classList.remove("loading")
            loading.innerHTML = `<p class=cross>&#x292C;</p>`
        }
    }
```

to 
```js
    async function formAction(action) {
        var data = inputToJson()
        if (data !== 0) { //don't run if there is an error in the data Json
            showChangeStatus("loading") // show loading div for status
            response = await fetch(`{{ basename }}/action/${action}`, {
                method: "POST",
                headers: {
                    "Content-Type": "application/json",
                },
                body: JSON.stringify(data), //post can only send data via strings
            }).then(saveStorage()) //save to storage if successful 
            showChangeStatus(response) //replace loading, show tick or cross
        }
        else {
            showChangeStatus("remove") //replace loading, show tick or cross with none 
        }
    }

    //function in control of status icons of post above
    function showChangeStatus(response) {
        var loading = document.getElementById("loader") //element showing status 
        if (response.status === "remove") { //show loading logo
            loading.innerHTML = "";
            loading.classList.remove("loading"); //append class with loading animation styling
        }
        else if (response.status === "loading") { //show loading logo
            loading.innerHTML = "";
            loading.classList.add("loading"); //append class with loading animation styling
        }
        else if (response.status === 201) { //then show a tick 
            loading.classList.remove("loading")
            loading.innerHTML = `<p class=tick>&#x2713;</p>`

        }
        else { //then show a cross 
            loading.classList.remove("loading")
            loading.innerHTML = `<p class=cross>&#x292C;</p>`
             document.getElementById("alert-text").textContent = "\r\n" + response.msg //or what the dict  response key is
             document.getElementById("alert").style.display = "block";
        }
    }
```

_Originally posted by @GeoDerp in https://github.com/davidusb-geek/emhass/issues/190#issuecomment-1936914644_

GeoDerp, 2024-02-11:
I guess the hardest thing I haven't worked out is how to get in the app log of the action instance

davidusb-geek, 2024-02-11:
I'm not certain but possibly some solutions here: https://stackoverflow.com/questions/66277372/writing-flask-console-logs-to-an-html-template-as-and-when-it-appears

GeoDerp, 2024-02-11:
Haven't looked into this. It may grab all logs. I think we want to get just the ones from app.logger although maybe we can filter before we send it out 🤔

closed 2024-03-03 by davidusb-geek