davidusb-geek/emhass #193: Related deferrable load parameters defaulting

GeoDerp, 2024-02-12:
There seems to be a lot of users settings parameters that should be the same amount as the derivable load and getting an error.

I had two ideas to improve this:

One: edit the utils to check each dict to see if they are the same length. If they aren't pad the dict with default (or the first item in dict) values

Two: in utils create an error log that states that there are missing values and can't continue unless you resolve them.

GeoDerp, 2024-02-12:
@davidusb-geek , may be good to give this a check

davidusb-geek, 2024-02-12:
So PR #194 implements the padding solution?
I this that is a good solution.
But I will change these defaults values:
```
params['optim_conf']['def_total_hours'].append(0)
params['optim_conf']['P_deferrable_nom'].append(0)
```
And most importantly put warning sign indicating that we are filling these with default values but the configuration should be fixed on the user side.

davidusb-geek, 2024-02-25:
I think that this is done

closed 2024-02-25 by davidusb-geek